FydeOS/chromium_os-raspberry_pi #108: [Issue report][Bug report]Cannot build_packages, emerge there are no ebuilds to satisfy "chromeos-base/chrome-icu-86.0.4240.28_rc for /build/rpi4"

PatrickChenHZ, 2020-11-27:
Scanning Configuration files...
Exiting: Nothing left to do; exiting. :)
13:23:07 INFO    : Running 'eclean -d packages' to clean up stale binpkgs
 * Building file list for packages cleaning...
 * Your packages directory was already clean.
13:23:07 INFO    : Elapsed time (update_chroot): 0m39s
13:23:07: INFO: Creating sysroot directories.
13:23:07: INFO: Installing configurations into sysroot.
13:23:07: INFO: Setting up portage in the sysroot.
13:23:08: INFO: Selecting profile: /mnt/host/source/src/overlays/overlay-rpi3/profiles/base for /build/rpi3
13:23:08: INFO: Updating toolchain.
13:23:08: INFO: Cross-compiler already up to date. Nothing to do.
13:23:08 INFO    : Clearing shadow utils lockfiles under /build/rpi3
13:23:09 INFO    : PORTAGE_BINHOST:  gs://chromeos-prebuilt/board/arm-generic/postsubmit-R86-13420.0.0-36875-8871426066622112112/packages
13:23:09 INFO    : Rebuilding Portage cache
13:23:11 INFO    : Checking package dependencies are correct: virtual/target-os virtual/target-os-dev virtual/target-os-factory virtual/target-os-factory-shim virtual/target-os-test
!!! CONFIG_PROTECT is empty for '/build/rpi3/'

These are the packages that would be merged, in order:

Calculating dependencies  . .. ... done!

emerge: there are no ebuilds to satisfy "~chromeos-base/chrome-icu-86.0.4240.28_rc" for /build/rpi3/.
(dependency required by "chromeos-base/chromeos-chrome-86.0.4240.28_rc-r1::fyde-for-rpi" [ebuild])
(dependency required by "virtual/chromeos-interface-1-r5::chromiumos[-chromeless_tty]" [ebuild])
(dependency required by "virtual/target-chromium-os-1-r143::chromiumos" [ebuild])
(dependency required by "virtual/target-os-1-r3::chromiumos" [ebuild])
(dependency required by "virtual/target-os" [argument])
13:23:17 ERROR   : emerge detected broken ebuilds. See error message above.
(cr) (release-R86-13421.B/(7bd3fbb...)) patrickchen@primaryVM ~/trunk/src/scripts $ 

Very sure that my overlay repo is R86-rpi4, my chromium source is chromium-m86,and my chromiumos source tree is release-R86-13421.B, They should match
I also failed to figure this out. I had tried everything recently. New discovery is that we have to use their chromium source, and must have it locally setup, you can refer to build.conf file to find this. But The error didn't change with local chromium source. I also tried following Emerge official doc to trouble shoot, and run emerge-rpi4 chromeos-base/chrome-icu This end up with another error complaning about cannot find package NSS. And Even I installed NSS and PKG_CONFIG manually in chroot, it still won't be able to find it. Another Attmpt I made is compiling chromium out side chroot, and hope it will also compile chrome-icu, but it end up not compiling and throw error"undefined symbol: chromeos::DBusClientFydeOS::DBusClientsFydeOS(bool)"

Considering repo is still being updated, I think the instruction is just too old, or author add something that need extra steps.

Please also refer to old ISSUE about facing error while building packages. There is a lot of people having this issue

PatrickChenHZ, 2020-11-27:
also my chromium source is cloned from Fydeos/chromium-raspberry-pi

yunshansimon, 2020-12-03:
Pls, update the latest overlays. If you use chromium src from us, you need to modify `chromeos-base/chromeos-chrome-86.0.4240.28_rc-r1::fyde-for-rpi` , try modify `~chromeos-base/chrome-icu-86.0.4240.28_rc` to `>=~chromeos-base/chrome-icu-86.0.4240.28_rc`.  You can also mask our chrome ebuild file by add `chromeos-base/chromeos-chrome::fyde-for-rpi` to `overlay-rpi4/profiles/base/package.mask`, build system will compile the ebuild in chromiumos instead.

PatrickChenHZ, 2020-12-04:
HI, I just follow the instruction and tried, with both, the build_package process ends up with "All ebuilds are masked", you need to unmusk at least....
What I tried is change ~chromeos-base/chrome-icu-${PV} to >=chromeos-base/chrome-icu-${PV}, though i don't understand why there needs an "~", but I when I removed it, it looks like it is compelling. 
Can you explan what is the purpose for ~ and what it will effect when I removed it. Thanks you so much.

yunshansimon, 2020-12-04:
It's my fault, you are right. `~` means need a specific version of the package.  My repo is a little early version. You will never pass the compiling if you haven't exactly some version of `chrome-icu` as mine.

PatrickChenHZ, 2020-12-06:
Thanks for your reply, my build process works fine now. The new problem I encountered is it will stuck at somepoint of build process and keep printing out jobs:791 of 989 complete,16running for the rest of the day. I did some debug work, and found it will stop at raspberrypi-kernel-5.4.74-r2. And won't pass that.
The log for kernel package building shows it cannot find /mnt/host/source/src/third_party/kernel/v5.4-rpi, then it will say cloning into bare repository /var/cache/chromeos-cache/distfiles/egit-src/kernel-rpi
But I found both directory is just empty.
I have no idea how to fix this issue, should I manually download Raspberrypi kernel source and put that into it's directory?

Thanks,
patrick chen

PatrickChenHZ, 2020-12-06:
updates, I found that the ebuild file for kernel has a place holder repo link, But what link and commit should I replace it to.
it currently says FydeOS-for-you-overlays.

PatrickChenHZ, 2020-12-06:
Sorry, my problem. It is necessary to setup a github ssh key in the host building system to clone the kernel repo, And i am using a VM for this build so it is not already setup.  I would recommend add this hint to the readme instruction, thanks

PatrickChenHZ, 2020-12-06:
I am not sure which stage am I suppose to have the git ssh key seted up so that kernel repo will clone automaticly, probably board initialize stage. My current make up solution is manually clone overlay repo branch fydeos-5.4 into /mnt/host/source/src/third_party/kernel
 and rename the repo folder to v5.4-rpi

PatrickChenHZ, 2020-12-06:
I had successfully built the chromium os, and also fiured out USB boot support to it, I will be adding some additional instructions to readme, and creat a pull request regarding instruction inprovement and issue fixing and usb boot, thanks to Mr.Yang so much.

Ruffio, 2021-01-05:
Shouldn't this issue be closed since it has been resolved?