FydeOS/chromium_os-raspberry_pi #109: [Issue report] chromiumos_image_r86r1-rpi4b not work HDMI audio on RPi4, but fixed

osakanataro, 2020-12-05:
**Describe the bug or issue**
chromiumos_image_r86r1-rpi4b not work HDMI audio on RPi4
But, I tried to change cmdlist.txt, then worked HDMI audio

**Describe your setup**
 - RPi 4B with 8GB RAM
 - chromiumos_image_r86r1-rpi4b 

**To Reproduce**
I add string "snd_bcm2835.enable_compat_alsa=1" to cmdlist.txt.

>chronos@localhost ~ $ sudo mount /dev/mmcblk0p12 /mnt
>chronos@localhost ~ $ cat /mnt/cmdline.txt
>coherent_pool=6M smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 console=serial0,115200 earlyprintk=serial0,115200 log_level=7 root=/dev/mmcblk0p3 rootfstype=ext2 rootwait ro cros_debug cros_secure elevator=deadline snd_bcm2835.enable_compat_alsa=1
>chronos@localhost ~ $ 

But worked only left microHDMI.

TDI4, 2020-12-05:
Same happened with me. Just flashed that build today, and no sound on HDMI, only analog output works. Where can I find _cmdlist.txt_?

osakanataro, 2020-12-06:
cmdlist.txt was found on mmcblk0p12 partition.
But this partition not mount.

1. login
2. Ctrl-Alt-T open a shell tab
3. type "shell" to run bash
4. mount EFI partition, "sudo mount /dev/mmcblk0p12 /mnt"
5. backup old cmdlist.txt, "sudo cp /mnt/cmdlist.txt"
6. edit cmdlist.txt "sudo vi /mnt/cmdlist.txt"

osakanataro, 2020-12-06:
how to check audio device.
on bash shell.
**execute "aplay -l"**
```
chronos@localhost ~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 3/3
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 2: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
chronos@localhost ~ $
```

My environment , left microHDMI is "card1, device0" 
play sound file, "aplay sound.wav -D hw:1,0"
```
chronos@localhost ~ $ aplay Downloads/dq.wav -D hw:1,0
Playing WAVE 'Downloads/dq.wav' : Unsigned 8 bit, Rate 11025 Hz, Mono
chronos@localhost ~ $
```

If you use headphone output, "aplay sound.wav -D hw:2,0"

nkhs9412235, 2020-12-09:
I have tried, but not works.
Could anyone help me？
[image]

osakanataro, 2020-12-09:
It looks your modify cmdlist.txt was failed.

my cmdlist.txt is follows
`coherent_pool=6M smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 console=serial0,115200 earlyprintk=serial0,115200 log_level=7 root=/dev/mmcblk0p3 rootfstype=ext2 rootwait ro cros_debug cros_secure elevator=deadline snd_bcm2835.enable_compat_alsa=1`

nkhs9412235, 2020-12-09:
I am so confused by cmdlist.txt or cmdline.txt.

[image]

osakanataro, 2020-12-10:
sorry, it's typo. filename is cmdline.txt

Fusion3Design, 2020-12-14:
this seems to work

osakanataro, 2020-12-17:
chromiumos_image_r86r2-rpi4b worked fine on left microHDMI port.
right microHDMI port is video only,