FydeOS/chromium_os-raspberry_pi #110: WiFi not working

sniffski, 2020-12-06:
I've just installed "chromiumos_image_r86r1-rpi4b.img.xz" on rpi400 and I'm not able to use WiFi. When I go to settings and try to connect manually to my home network, it says unknown error. Anyone having the same issue? Suggestions on how to fix this will be perfect... :)

yunshansimon, 2020-12-06:
Our rpi400 is still on the way. You can copy the fold `/lib/firmware`  from the official image to the Chromium OS partition 3.

sniffski, 2020-12-06:
Can you just provide me with link to official image... Not sure which one you are talking about.
Isn't it rpi400 same as rpi4 with keyboard?

sniffski, 2020-12-07:
Hope you mean "2020-12-02-raspios-buster-armhf"... I'll try with that one... I'm preparing this rpi400 for my granny who is 92, I don't know how much time do I have to complete this... :D
I'm not joking! :D

sniffski, 2020-12-07:
Man it really worked!!! Thank you! Thank you! Thank you! :+1: :trophy: :dancing_men:

sniffski, 2020-12-07:
I don't see anything in Audio settings, I guess that's not working as well. Is there anything I can quickly work around like I did for WiFi? 
[image]

yunshansimon, 2020-12-07:
If there is no sound over HDMI, you may try [hdmi issue](https://github.com/FydeOS/chromium_os-raspberry_pi/issues/109) as well.

sniffski, 2020-12-09:
Thanks again! That worked as well!

closed 2020-12-12 by fydeos-alpha

seghier, 2021-01-30:
I have the same problem, i install FydeOS in raspberry 3b+ and when i connect it to screen and in wifi step can't connect it , just tell me there is an error

OMProduction, 2023-07-15:
Still kind of confused where can I find the image of chromium I used and the /lib/firmware 

I am mostly a beginner only know a few things about linux