FydeOS/chromium_os-raspberry_pi #111: [Issue report]

deathblade666, 2020-12-06:
**Describe the bug or issue**
total storage is displaying as 4GB 2.1GB is being used out of the box.

**Describe your setup**
 - Raspberry pi 4 B 4GB, 32GB sdcard
 - ChromiumOS r86 r1
 - N/A

**To Reproduce**
Steps to reproduce the behavior:
1. check Storage management in settings

**Expected behaviour**
Show full sdcard size (minus system and out of box files)

**Screenshots and/or hardware setup photos**
If applicable, add screenshots to help explain your problem. If your problem is hardware related, it's much easier to include a photo of how your setup looks like.

**Additional context**
Due to this I am unable to enable Linux(beta)

PatrickChenHZ, 2020-12-06:
You should manually expand partition. I suspect there is not an auto repartition script in chromium pi like offcial rpi images does.  Gparted or anything similar on windows or mac should do.

yunshansimon, 2020-12-07:
You can do it manually. Try this command in the shell: 
`sudo  /usr/sbin/expand-partition.sh -d $(findmnt -n -o source /mnt/stateful_partition)`
Or 
`sudo rm /mnt/stateful_partition/.autoexpanded` The system will try expand the partition on next booting.

deathblade666, 2020-12-08:
Thanks guys,

That did it.

techie59, 2020-12-22:
Hello,
Once installed, I go to the terminal but the ChromeOS says that it does not know the sudo command. Do I do something wrong?

techie59, 2020-12-22:
> You can do it manually. Try this command in the shell:
> `sudo /usr/sbin/expand-partition.sh -d $(findmnt -n -o source /mnt/stateful_partition)`
> Or
> `sudo rm /mnt/stateful_partition/.autoexpanded` The system will try expand the partition on next booting.

Hello,
Under which application in Chrome OS do you type this command?
Thanks,

fydeos-alpha, 2020-12-22:
> Under which application in Chrome OS do you type this command?

In Chromium OS desktop press `ctrl` + `alt` + `t` and then type `shell` to invoke web-based bash shell

Ruffio, 2021-01-05:
Shouldn't this issue be closed since it has been resolved?

fydeos-alpha, 2021-01-05:
> Shouldn't this issue be closed since it has been resolved?

Leaving these common issues here in a more prominent place to help beginners..

Ruffio, 2021-01-05:
Is it common to close resolved issues. It is better to update the readme page with relevant FAQ/informations.

HawkCollective2890, 2022-05-02:
I ran power wash and it corrected it.

locness3, 2022-08-20:
> Or `sudo rm /mnt/stateful_partition/.autoexpanded` The system will try expand the partition on next booting.

Is this file added on first boot after attemtping to auto-expand, or is it already present? If it is the first, why would the auto-expand complete without succeeding?

locness3, 2022-08-20:
> Is it common to close resolved issues. It is better to update the readme page with relevant FAQ/informations.

+1. Issue tickets are not designed to work as searchable documentation.