FydeOS/chromium_os-raspberry_pi #113: Overclock

thealex160, 2020-12-11:
Hello!

Where is the location of the config.txt? Looking to overclock the PI a bit.
Great job on the OS! Very happy with it, thanks!

thealex160, 2020-12-12:
Having actually located the config.txt, it does not contain any obvious cpu/gpu speed info like it normally does on Pi. Can it be added, or does chromium os need a different code/use a different file for this?

fydeos-alpha, 2020-12-12:
we have not tested overclocking, but you can try

h0rn3t, 2020-12-27:
@thealex160  where is located config.txt?

Ruffio, 2021-01-05:
@fydeos-alpha does the config.txt support the normal Pi settings: https://www.raspberrypi.org/documentation/configuration/config-txt/ ?

Heclalava, 2021-01-22:
I am also trying to locate the config.txt file. Linux Mint will only mount ROOT-A, STATE, and OEM partitions. I am not finding config.txt in any of these. I will try from terminal to see if I can find it.

fydeos-alpha, 2021-02-20:
@Heclalava   does overclocking work?

Heclalava, 2021-02-20:
@fydeos-alpha yes it does, but without the ability to root the Android subsystem I gave up on this OS a long time ago. It's like trying to use Linux with without sudo.

crux161, 2024-03-07:
> @fydeos-alpha yes it does, but without the ability to root the Android subsystem I gave up on this OS a long time ago. It's like trying to use Linux without sudo.

ummm… Please (kindly) Read the manual 😬

Heclalava, 2024-03-08:
> > @fydeos-alpha yes it does, but without the ability to root the Android subsystem I gave up on this OS a long time ago. It's like trying to use Linux without sudo.
> 
> ummm… Please (kindly) Read the manual 😬
> 
> `https://fydeos.io/docs/knowledge-base/getting-started/shell-access/`

Maybe now, but at the time 3 yeas ago, I couldn't get it rooted.

crux161, 2024-03-08:
I realize there was a substantial change to the security model for fydeos and your comment was about dealing with that older model. Sorry about that. 

I stumbled down this path and found what I needed so hopefully someone else looking for the same can gain root and lock things down for themselves. 

`mount -o remount rw /` (as root) from the fydeos console (not the android subsystem like I previously linked to sorry 😅)

then `passwd root` etc and it should commit the change to root’s credentials giving you the full control over your system that a real user deserves. Without it, you’re absolutely right — fortunately there’s an obscure path to victory here. 

Hope this helps. Sorry again