FydeOS/chromium_os-raspberry_pi #114: unkown admin restrict and problem of changing places of applications in launcher

kenryuS, 2020-12-16:
**Describe the bug or issue**
I have the chromium os with an unknown admin that is blocked: "connect android phone" and entering developer mode on the first boot. It came from the first boot. It has only one account registered. Also, the placement of the applications on the launcher forcibly resets/changed every time I stop change the placement/reopen launcher.

**Describe your setup**
 - Raspberry Pi 4B with 4GB RAM
 - Your Chromium OS for Raspberry Pi release version, e.g. Chromium OS r86.04240.186 (32-bit) r4b

**To Reproduce**
Steps to reproduce the behavior for admin blocking:
1. Go to 'Settings'
2. Click on 'Connected devices' / Scroll down to 'Connected devices'
4. See you can't turn on the Android Phone column.
Steps to reproduce the behavior of resetting of placement of applications on launcher.
1. drag the application(s) to another place.
2. wait or reopen the launcher
3. see if the placement of application(s) reset/changed.

**Screenshots and/or hardware setup photos**
[Pictures and explanations of the problems.zip](https://github.com/FydeOS/chromium_os-raspberry_pi/files/5699877/Pictures.and.explanations.of.the.problems.zip)
Please open this zip file.

**Additional context**
So basically, you must debug:
1. delete the administer setting
2. debug the code on change the application placement code on the launcher(so applications stay in the place where you drag)

fydeos-alpha, 2020-12-16:
“Connected device” feature isn't available in vanilla Chromium OS, unfortunately, like many other items. Google Assistant, for example, is another one of those.

There are no "admin restrictions" per-se, if there was any, they'd be shown up in <chrome://policy>

You will see the hollowed, non-functional frontend of those unavailable items if the chromium project is unchanged from upstream. However they really bother you that much, it's easy to hide them by commenting out the relevant sections in chromium and rebuild.

closed 2020-12-16 by fydeos-alpha

fydeos-alpha, 2020-12-16:
I am unable to reproduce your reported issue 2 about changing the order of the applications. In our test kits, if you drag and place any app from posA to posB, they stay there.

Closing this issue now, further comments welcomed.