FydeOS/chromium_os-raspberry_pi #125: [Issue report] Failed to emerge chromeos-base/libwidevine-0.0.1

cjbd, 2020-12-23:
**Describe the bug or issue**
Hello, I followed the steps in the README.md, and made it to `build_packages` step, most packages passed, there're about 11 left, I ran a few times, it always errors out at below:
```sh
./build_packages --board=rpi4 --nowithautotest
```
**Failed to emerge chromeos-base/libwidevine-0.0.1**

```
libwidevine-0.0.1: 20:41:20 >>> Install chromeos-base/libwidevine-0.0.1 into /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/image/
libwidevine-0.0.1: !!! doexe: /var/cache/chromeos-chrome/chrome-src/src/out_rpi4/Release/libwidevinecdm.so does not exist
libwidevine-0.0.1:  * ERROR: chromeos-base/libwidevine-0.0.1::cros-pi failed (install phase):
libwidevine-0.0.1:  *   doexe failed
libwidevine-0.0.1:  * 
libwidevine-0.0.1:  * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:libwidevine-0.0.1:20201223-124119.log
libwidevine-0.0.1:  * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/temp/build.log
libwidevine-0.0.1:  * CWD: /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work
libwidevine-0.0.1:  * S:   /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work
libwidevine-0.0.1:  * QA Notice: file does not exist:
libwidevine-0.0.1:  * 
libwidevine-0.0.1:  * 	doexe: /var/cache/chromeos-chrome/chrome-src/src/out_rpi4/Release/libwidevinecdm.so does not exist
>>> 20:41:21 Jobs: 3 of 11 complete, 1 failed                Load avg: 1.70, 2.55, 6.25

 * Messages for package chromeos-base/libwidevine-0.0.1 merged to /build/rpi4/:
 * Log file: /build/rpi4/tmp/portage/logs/chromeos-base:libwidevine-0.0.1:20201223-124119.log

 * ERROR: chromeos-base/libwidevine-0.0.1::cros-pi failed (install phase):
 *   doexe failed
 * 
 * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:libwidevine-0.0.1:20201223-124119.log
 * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/temp/build.log
 * CWD: /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work
 * S:   /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work
20:41:21 ERROR   : 2020年 12月 23日 星期三 20:41:21 HKT
20:41:21 ERROR   :  PGID  PPID   PID     ELAPSED     TIME %CPU COMMAND
20:41:21 ERROR   :     8     4     8    09:57:50 00:00:00  0.0 -bash
20:41:21 ERROR   : 31174     8 31174       01:51 00:00:00  0.0  \_ /bin/bash ./build_packages --board=rpi4 --nowithautotest
20:41:21 ERROR   : 31174 31174  3045       00:00 00:00:00  0.0      \_ /bin/bash ./build_packages --board=rpi4 --nowithautotest
20:41:21 ERROR   : 31174  3045  3046       00:00 00:00:00  0.0          \_ ps f -o pgid,ppid,pid,etime,cputime,%cpu,command
20:41:21 ERROR   : Arguments of 31174: ./build_packages '--board=rpi4' '--nowithautotest'
20:41:21 ERROR   : Backtrace:  (most recent call is last)
20:41:21 ERROR   :  build_packages:462:main(), called: die_err_trap  
20:41:21 ERROR   : 
20:41:21 ERROR   : Command failed:
20:41:21 ERROR   :   Command '( if [[ "${FLAGS_run_goma}" -eq "${FLAGS_TRUE}" ]]; then
20:41:21 ERROR   :     info "Starting goma compiler_proxy."; goma_ctl="${GOMA_DIR:-${HOME}/goma}/goma_ctl.py"; "${goma_ctl}" restart; trap "'${goma_ctl}' stop" EXIT;
20:41:21 ERROR   : fi; sudo -E "${EMERGE_CMD[@]}" "${EMERGE_FLAGS[@]}" "${PACKAGES[@]}" --useoldpkg-atoms="${CRITICAL_SDK_PACKAGES[*]}" --rebuild-exclude="${CRITICAL_SDK_PACKAGES[*]}" )' exited with nonzero code: 1

```

**Describe your setup**
 - Your Raspberry Pi model and spec variant, e.g. 4B with 2GB RAM
   I haven't completed building, no device used yet
 - Your Chromium OS for Raspberry Pi release version, e.g. Chromium OS r77 r3
   same version in README.md, m86
 - Your hardware configuration and setup, e.g. how you have wired the Pi board with any other peripheral connected to it
   I haven't completed building, no device used yet

**To Reproduce**
Steps to reproduce the behavior:
1. cros_sdk to enter chroot
2. run `./build_packages --board=rpi4 --nowithautotest` command
3. check shell output

**Expected behaviour**
expect `build_packages` pass all steps

**Screenshots and/or hardware setup photos**
no screenshot, pls check below log

**Additional context**
logfile: /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/temp/build.log

```
 * Package:    chromeos-base/libwidevine-0.0.1
 * Repository: cros-pi
 * USE:        arm elibc_glibc kernel_linux userland_GNU
 * FEATURES:   network-sandbox splitdebug userpriv
 * Running stacked hooks for pre_pkg_setup
 *    sysroot_build_bin_dir ...
 [ ok ]
 * Running stacked hooks for post_pkg_setup
 *    python_eclass_hack ...
 [ ok ]
20:41:19 >>> Unpacking source...
20:41:19 >>> Source unpacked in /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work
 * Running stacked hooks for post_src_unpack
 *    asan_init ...
 [ ok ]
 * Running stacked hooks for pre_src_prepare
 *    build_toolchain_catch ...
 [ ok ]
20:41:19 >>> Preparing source in /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work ...
20:41:19 >>> Source prepared.
20:41:20 >>> Configuring source in /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work ...
20:41:20 >>> Source configured.
20:41:20 >>> Compiling source in /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work ...
20:41:20 >>> Source compiled.
 * Running stacked hooks for post_src_compile
 *    python_eclass_hack ...
 [ ok ]
20:41:20 >>> Test phase [not enabled]: chromeos-base/libwidevine-0.0.1
 * Running stacked hooks for pre_src_install
 *    tidy_setup ...
 [ ok ]
20:41:20
20:41:20 >>> Install chromeos-base/libwidevine-0.0.1 into /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/image/
!!! doexe: /var/cache/chromeos-chrome/chrome-src/src/out_rpi4/Release/libwidevinecdm.so does not exist
 * ERROR: chromeos-base/libwidevine-0.0.1::cros-pi failed (install phase):
 *   doexe failed
 * 
 * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:libwidevine-0.0.1:20201223-124119.log
 * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/temp/build.log
 * CWD: /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work
 * S:   /build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/work
[33;01m * [39;49;00mQA Notice: file does not exist:
[33;01m * [39;49;00m
[33;01m * [39;49;00m	doexe: /var/cache/chromeos-chrome/chrome-src/src/out_rpi4/Release/libwidevinecdm.so does not exist
```

fydeos-alpha, 2020-12-23:
Assuming you are building chromium as well from [our repo](https://github.com/FydeOS/chromium-raspberry_pi).

You need to check your `chrome-root` under `${chrome-root}/src/third_party/widevine/cdm/chromeos/arm` and if libwidevinecdm.so exist. If not, rerun `gclient sync` or figure out a way to make sure the missing file is there.

cjbd, 2020-12-24:
@fydeos-alpha, thank you for the update!

yes, i'm building chromium as well, but i'm using vanilla chromium, it's fetched from google

i checked `${chrome-root}/src/third_party/widevine/cdm/` using both `Files` from ubuntu os and `chroot`, there's no `chromeos/arm`, but all other folders don't have that either ...

i noticed that some files under `/build/rpi4/tmp/portage/chromeos-base/libwidevine-0.0.1/` are broken, and `libwidevine-0.0.1` is the only one in `/build/rpi4/tmp/portage/chromeos-base`

[image]

is it possible that the temp files for libwidevine are broken? from log i can see, the build was completed, it's saying install failed ...

fydeos-alpha, 2020-12-24:
vanilla chromium doesn't have widevinecdm support, to pass the build, you need to remove widevine from the relevant ebuild files. 

Or consider using [chromium-raspberry_pi](https://github.com/FydeOS/chromium-raspberry_pi)

cjbd, 2020-12-24:
@fydeos-alpha, oh...
since i've started building with vanilla chromium, how do i remove widevine from ebuild files?

fydeos-alpha, 2020-12-24:
It's registered in `fyde-packages-0.0.1.ebuild` file, remove/comment the `widevine` package under `RDEPEND`as well as from the `IUSE`

cjbd, 2020-12-24:
@fydeos-alpha, sorry, i only found `chromeos-chrome-86.0.4240.260_rc-r1.ebuild`, where do i find `fyde-packages-0.0.1.ebuild`? 

i tried seaching using `Files` in the `chroot.img`, the closest file i can find is `/chroot/var/cache/edb/dep/mnt/host/source/src/overlays/project-cros-pi/virtual/fyde-packages-0.0.1`

```
DEFINED_PHASES=-
DEPEND=chromeos-base/auto-expand-partition chromeos-base/power_wash_command chromeos-base/fyde-chromedev-flags widevine? ( chromeos-base/libwidevine ) kiosk_demo? ( chromeos-base/fyde-kiosk-demo ) fyde_extension? ( chromeos-base/fyde-shell-daemon-bin )
DESCRIPTION=empty project
EAPI=5
HOMEPAGE=https://fydeos.io
IUSE=kiosk_demo widevine fyde_extension
KEYWORDS=*
LICENSE=BSD
RDEPEND=chromeos-base/auto-expand-partition chromeos-base/power_wash_command chromeos-base/fyde-chromedev-flags widevine? ( chromeos-base/libwidevine ) kiosk_demo? ( chromeos-base/fyde-kiosk-demo ) fyde_extension? ( chromeos-base/fyde-shell-daemon-bin )
SLOT=0
_md5_=c505da669389cb53e51dde188c33e4c8
```

is this the one to edit?

fydeos-alpha, 2020-12-24:
all files that I was referring to belonged to [this current project](https://github.com/FydeOS/chromium_os-raspberry_pi), it'd be helpful if you have followed the Guide and have them located somewhere outside cros_sdk repo and created symlinks and bind mounts.

search this git repository rather than the `choot.img`.

cjbd, 2020-12-24:
@fydeos-alpha, thanks a lot!
i thought they must be files inside chroot.img, i copied this project to overlay folder directly
i edited the 2 files, the build_packages passed!

pls keep this ticket open for a while, i'm going for next step

cjbd, 2020-12-24:
@fydeos-alpha, last step went wrong, the card is new, please help, thanks a lot! 

```
(cr) (release-R86-13421.B/(7bd3fbb...)) pi@dev ~/trunk/src/scripts $ cros flash usb:// rpi4/latest
Removable device(s) found. Please select/confirm to continue:
  [0]: Generic Mass Storage Device 14.9G (/dev/sdc)
Please choose an option [0-0]: 
Input is not an integer
Please choose an option [0-0]: 0
 [###############################################################################################] 100%
The backup GPT table is not on the end of the device. This problem will be corrected by write.
Checking that no-one is using this disk right now ... FAILED

This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.

sfdisk: Use the --force flag to overrule all checks.
partx: /dev/sdc: error updating partition 1
14:14:18: ERROR: Failed copying image to device /dev/sdc
14:14:18: NOTICE: cros flash completed successfully.
```

fydeos-alpha, 2020-12-24:
the image should work afaik, try it out.

cjbd, 2020-12-24:
@fydeos-alpha , thanks a lot! the image works, it's booted
sorry to bother you over and over again

when i boot from the image, this error message is in the top screen:

```
You will not be able to add users because Google API keys are missing. See
https://www.chromium.org/developers/how-tos/api-keys for details.
```

but i did follow the README.md to add the key in `~/.googleapikeys`, is this location in host or chroot? i put in host, btw i do remember see log in build_packages saying they found google key

also, after picking wifi, it stuck in this almost blank window, saying `Please wait...`, and have been spinning long time ...

[image]

shall i wait or what?

fydeos-alpha, 2020-12-24:
> but i did follow the README.md to add the key in ~/.googleapikeys, is this location in host or chroot? i put in host, btw i do remember see log in build_packages saying they found google key

`.googleapikeys` in the host is fine, it gets cloned to chroot too so you should really see it in both places, especially if you remember the log confirmed it was found.

The WSOD is more than likely caused by API issues causing the OS being unable to talk to Google's login services.

Not sure how you've configured and applied API accesses from your GCP account, unfortunately it's out of the scope of this project, can't help you too much on this front.

Comments welcome, issue closed.

closed 2020-12-24 by fydeos-alpha

cjbd, 2020-12-24:
@fydeos-alpha

i would like to take this opportunity to express my gratitude to this project, you guys rock!

it's been a very exciting journey to build an entire OS from scratch and be able to use it in raspberry pi!

Awesome job, FydeOS team!