FydeOS/chromium_os-raspberry_pi #126: Unable to enable/install Play Store (I'm outside of china)

andybe, 2020-12-29:
While I know Android is working on the pi 4 and I'm in EU. There is no common thing why I should't install Googel PlayStore applications on a rasberry pi 4.

Please make an description.

fydeos-alpha, 2020-12-30:
the project is for Chromium OS. Not Chrome OS, not FydeOS either. 

Chromium OS does not support Android subsystem, it doesn't concern where you are.

closed 2020-12-30 by fydeos-alpha

andybe, 2020-12-30:
sorry  @fydeos-alpha

fydeos-alpha, 2021-01-05:
> sorry @fydeos-alpha
> 
> hm ... https://github.com/ " FydeOS " /chromium_os-raspberry_p

It suggests that this project is maintained by the FydeOS team, per the README.

andybe, 2021-01-05:
@fydeos-alpha 

In the official documentation it's written,- found it yesterday
Two nice lines for pi users would be fine:

- Widevine isn't include (most user expected - Kodi etc.)
- Google Appstore not included.

With the reference to the official documentation.