FydeOS/chromium_os-raspberry_pi #128: is it safe to log in with my google account?

Itai-Nelken, 2021-01-02:


fydeos-alpha, 2021-01-03:
please define "safe"?

fydeos-alpha, 2021-01-03:
not an issue, closed here. further comment welcome.

closed 2021-01-03 by fydeos-alpha

Itai-Nelken, 2021-01-03:
@fydeos-alpha I remember reading somewhere that you shouldn't log in with your google account, but it might have been a old chromiumOS test image. so I just wanted to be sure.

fydeos-alpha, 2021-01-03:
Our previous "test" image has a well-known ssh key installed in the system for faster remote connection in the shell level. The warning message regarding this was about shell level credentials, nothing concerning the frontend.

Beginning with the recent releases we don't do test image anymore, you will have to set your own shell user password if you wish to connect to the OS via ssh.

Itai-Nelken, 2021-01-03:
@fydeos-alpha thanks a lot!