FydeOS/chromium_os-raspberry_pi #132: Intermittent Pauses in inputs

rpangrazio, 2021-01-07:
Version: Version 86.0.4240.186 (Official Build) (32-bit)
Device: Rasp Pi 4 4GB Ram 32GB SD(expanded to fill all of the card)
Browser Extensions: Grammarly, Offline Good Docs

Problem:
Every few minutes the system stops responding to inputs. I've seen this on both USB mouse/keyboard and Bluetooth mouse/keyboard. In both cases, devices are separate devices, not combined. 

The system as a whole does not appear to freeze as if I do it when I see a cursor, it still blinks.

rpangrazio, 2021-01-07:
Update:
At least part of the GUI may not update. I just had my mouse hoving over the close X and moved it a second before a freeze started and the Close tooltip stayed up till the freeze ended.

fydeos-alpha, 2021-01-07:
it sounds like a graphics mode induced glitch. With the r86r2 release we are using `fkms` mode which kernel selects the screen mode for the screen but the graphics firmware manages the video output path. 

In full `kms` mode the kernel manages everything but it introduces incompatibilities with the Pi3 models.

In full `kms` mode glitches like this could be improved to some degree, we may consider to produce an image just for Raspberry Pi 4 models with graphics in `kms` mode.

rpangrazio, 2021-01-07:
I do have it hooked up to a TV. Does that present any problems? Should I move to a monitor that will allow it to support different resolutions?

cheyrou23, 2021-01-13:
I've seen the same issue:

- RPi4 4GB
- chromiumos_image_r86r2-rpi4b.img
- separately wired USB mouse and keyboard
- HDMI to VGA cable to generic PC monitor

Pauses for maybe 10sec every few minutes then recovers.

rpangrazio, 2021-04-21:
It looks to be corrected with the latest release. I am continuing to test now. It seemed to be OK on my TV setup. I did get some screen artifacts, but that is not being replicated on a computer moniter. If I keep seeing them I'll get a screenshot and open a separate ticket as I'm not convinced they are related.

rpangrazio, 2021-04-30:
I'm closing as I don't have the same problem on the newest build

closed 2021-04-30 by rpangrazio