FydeOS/chromium_os-raspberry_pi #134: Intermittent restarts

alexskinner, 2021-01-14:
Hi, 

I have a raspberry PI 8gb and 4gb and have tried the very latest release. What I see is intermittent restarts of the PI. Not necessarily when its doing something more commonly at Idle it will just restart.

I thought it might be under powered but I've trie the official raspberry PI PSU too.
I thought it could be thermal but its not overclocked and have tried different cases with active cooling. 

Are there any log files I could get at that might give a clue ?

Thanks

Alex

fydeos-alpha, 2021-01-15:
We've seen restarts occasionally but with our cases, they were most likely due to memory issues as we were tormenting the Chromium with WebGL intensive stuff. If yours happened during idle, I would still guess it's power. We don't use PI PSU on our testbed but a smartphone charger that supports high watts fast-charge. Perhaps you could try using an alternative power source and check if things could get better.

i/o related error could be another reason, so perhaps a new SD card would help.

To check log you'd need to establish a serial connection to the Pi to check dmesg or SSH to it and tail the `/var/log/messages` and check if there is any clue.

cjbd, 2021-01-15:
i experienced the same, then i found in release note, `sleep` is not supported yet
default setting in `device->power->when idl->while charging` is `sleep`
if you change that to something else, it should be fine

alexskinner, 2021-01-18:
Yes it turned out to be the same thing for me i'd missed the bit about
installing the always on extension

A

closed 2021-01-18 by fydeos-alpha