FydeOS/chromium_os-raspberry_pi #142: Getting 5GHz WiFi working on RPi4 in FydeOS

lmakarov, 2021-02-02:
**Describe the bug or issue**

I'm unable to connect to a 5GHz access point (in US) on RPi4 under FydeOS.

Only the lower band 5GHz channels (36-48) are discoverable.
Higher band channels (149-165) are not discoverable (checked with the WiFi Analyzer Android app).

**Describe your setup**

 - RPi4 8GB
 - FydeOS v11.4
 - FydeOS installed and booted from a Samsung T5 drive connected to USB 3.0 port

**To Reproduce**

Steps to reproduce the behavior:
1. Install the latest version of FydeOS (v11.4)
2. Attempt to connect to a 5GHz access point - fail

**Expected behaviour**

RPi4 can connect to a 5GHz WiFi network.

**Additional information**

From https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md:

> On the Raspberry Pi 3B+ and Raspberry Pi 4B, you will also need to set the country code, so that the 5GHz networking can choose the correct frequency bands.

This is normally done via `raspi-config` or `wpa_supplicant.conf` in Raspberry Pi OS.

I have not found a way to check or override the country code value in FydeOS.

fydeos-alpha, 2021-02-20:
it's a known issue that the Pi4 / Pi 400 kit does not have the most capable wifi module, the wifi firmware that we use also does not offer `raspi-config` or similar way to tweak the config. 

Based on our experience, the 5GHz wifi does not immediately show up, it usually turns up after we connect to a 2GHz. We also found that even if the Pi is connected to a 5GHz wifi, the actual transmission speed does not offer too much visible difference than the 2GHz one, probably due to the CPU limitation ...

lmakarov, 2021-02-25:
On Linux (Raspberry OS) with the country code properly set, there is a substantial difference between 2GHz and 5GHz WiFi performance on RPi4. In my case, it's an order of magnitude (10x) difference.

```
5GHz: 56.41 / 25.15 mbps (down/up)
2.4GHz: 6.17 / 2.64mbps (down/up)
```

Without the country code set, RPi4 may or may not see a 5GHz AP, may or may not connect to the 5GHz AP it can see, and may or may not perform well even if connected.

> the wifi firmware that we use also does not offer raspi-config or similar way to tweak the config.

Could you elaborate a bit more on this?

lmakarov, 2021-02-25:
Also, to be fair, the same issue applies to various Android builds for RPi4.
Same with any Linux distro until the wifi country code is configured. With Linux I can set the country code to fix the issue. With Chromium OS / Android, I was not able to find any way to do this.

lmakarov, 2021-02-26:
I updated the firmware on my WiFi AP, which also led to a complete factory reset of the AP. I decided to leave all of the factory settings as-is. The AP automatically picked channel 36 for the 5GHz band (used to be 161).

This time, RPi4 (running the latest Chromium OS build from this repo) was able to connect to the 5GHz network! 🎉 
This did not work yesterday, when I manually forced channel 36 and tried to connect.

I looked into the AP settings, trying to find the one that did the trick in this case.

It's was the wireless mode:

- `802.11n/ac mixed` (RPi4 can connect)
- `802.11ac only` (RPi4 won't connect)

Comparing 802.11n 5GHz vs 802.11n 2.4GHz performance:

```
5GHz: 43.17 / 48.26 mbps (down/up)
2.4GHz: 7.20 / 7.34 mbps (down/up)
```

---

So my workaround for the time being to get 5GHz wifi to work:

- Allow `802.11n/ac mixed` mode
- Manually set a lower band channel (36-48)

---

The issue with supporting `802.11ac` (and higher band 5GHz channels) still stands true.
It will require finding a way to correctly set the wifi regulatory domain / country code.

closed 2021-04-17 by fydeos-alpha

lmakarov, 2021-04-21:
> fydeos-alpha added the "solved & solution provided" label 5 days ago

For anyone else landing here, a quick TL;DR:

- The issue with supporting `802.11ac` (as well as `802.11n` higher band 5GHz channels) in FydeOs still stands true.
- The issue is not limited to FydeOS. It affect any Chromium and/or Android based OS build for RPi4 out there.
- Solving it would require finding a way to correctly set the wifi regulatory domain / country code for the RPi4's WiFi module. It's a driver/configuration problem (no `raspi-config` for Chromium/Android).