FydeOS/chromium_os-raspberry_pi #145: Dual monitor on 400 broken

janat08, 2021-02-09:
Dual monitor on 400 image (1gb+) is broken. Setting hdmi safe makes resolution go to the lowest possible value.

mbodomi, 2021-02-11:
I dont want to open other issue
I have Pi400
flashed this image FydeOS_for_you_Pi400_v11.4_SP2.img.xz =) 
This image and sadly it has a diagonally streched screen on my monitor setup
How do I fix it?

mbodomi, 2021-02-11:
I have a 720p BENQ LCD with DVI

janat08, 2021-02-11:
Yea more specifically the second monitor ranges from being recognized by OS, but giving blank screen on one of the two (I think), to entirely ignoring the second one. I have junk monitors from 2000s.

mbodomi, 2021-02-11:
So you are using hdmi->vga converters?

janat08, 2021-02-11:
Yes

goalhelp, 2021-08-29:
> I dont want to open other issue
> I have Pi400
> flashed this image: FydeOS_for_you_Pi400_v11.4_SP2.img.xz =)
> ;; and sadly it has a diagonally streched screen on my monitor setup
> How do I fix it?

you can fix this issue with this 
hdmi_ignore_edid=0xa5000080

edit the below config accordig to your monitor or tv
hdmi_group=2
hdmi_mode=85
hdmi_force_plug=1

closed 2021-12-10 by fydeos-alpha