FydeOS/chromium_os-raspberry_pi #149: How to use the kiosk mode on the newest 86r2

FrederikSch, 2021-03-02:
Hey,

I really tried to use your description how to do it, but it wont work. 

Can you give like a little tutorial how to add a kiosk mode to the image?

greetings

DanielDestino, 2021-03-17:
Yes, was also unable to get Kiosk mode working on latest build.

yunshansimon, 2021-03-17:
There is a "kiosk demo" in overlay. Just rebuild all packages with USE="${USE} kiosk-mode". The google official "kiosk-mode" is a function that needs to pay money to Google, pls ref:  [support.google.com](https://support.google.com/chrome/a/answer/9273974?hl=en)

DanielDestino, 2021-03-17:
There is a alternative metod, with pressing Ctrl+Alt+K to activate Kiosk mode without management that was disabled for newer devices (Still works on older Chrome devices). If that code is still present it would be great if that could be activated again as default.

https://gist.github.com/tylermakin/c3136bc4ad4c66bc2b8f263410aa2464

DanielDestino, 2021-03-23:
Code updates related to that function in origional Chromium:
https://codereview.chromium.org/2548603004/#ps60001

In: chrome/browser/chromeos/login/ui/webui_login_view.cc

+  if (KioskAppManager::IsConsumerKioskEnabled()) {
+    accel_map_[ui::Accelerator(ui::VKEY_K,
+                               ui::EF_CONTROL_DOWN \| ui::EF_ALT_DOWN)] =
+        kAccelNameKioskEnable;
+  }

Unable to find it in newer builds.

ifzl, 2021-04-10:
Can someone explain (step by step)  how to enable the Kiosk mode ?

ifzl, 2021-04-12:
> 
> 
> Hey,
> 
> I really tried to use your description how to do it, but it wont work.
> 
> Can you give like a little tutorial how to add a kiosk mode to the image?
> 
> greetings

I have managed to do the following steps but still something is wrong or missing;
Step 1 =  CRTL+ALT+T  - this will open the Crosh window
Step 2 =  use the command:   shell     ( this will open the shell ) 
Step 3 =  use the command:   sudo mount -o remount rw /    ( this will give you the writing rights ) 
Step 4 =  use the command:   sudo vi /etc/chrome_dev.conf    ( this will bring you to the file you want to edit ) 
Step 5 =  add the line:    --force-kiosk-mode
Step 6 =  press Esc
Step 7 =  type:       :wq!    ( and then press Enter)
Step 8 =  sudo reboot 

So far this has not given me the result I am looking for.  Can someone indicate what I should change ?

fydeos-alpha, 2021-04-15:
Read the latest release note, it may offer some help.

closed 2021-04-15 by fydeos-alpha