FydeOS/chromium_os-raspberry_pi #150: Bluetooth audio not working correctly

hgpuke, 2021-03-08:
When playing anything from the browser, using a bluetooth connected speaker, the sound is very choppy to the extent that it is unusable.

Setup:
Raspberry Pi 4B with 4GB of RAM
FydeOS 11.4 
Chromium: Version 86.0.4240.260  (32 bitar)
Plattform: 13421.23.11.25 (Release Build v11.4 - sp2) stable-channel rpi4-fydeos
Kanal: beta-channel
Firmwareversion: 
ARC aktiverat: true
ARC: 6104653
Med företagsregistrering: false
Utvecklarläge: true

Keyboard and touchpad from Trust connected via wireless transmitter.
Screen connected to HDMI1 port in 1080p
FydeOS running off a Samsung 64GB memory stick

To Reproduce
Steps to reproduce the behavior:
0. Connect the bluetooth speaker using settings in FydeOS
1. Go to youtube.com
2. Click on any video
3. Start playing
4. Listen to the sound from the bluetooth speaker. It is horrible and any dialogue can not be understood.

Expected behaviour
Bluetooth audio should be as clear as audio through the jack.

Additional context
Lowering the resoultion on the video has no effect

uwence, 2021-03-11:
https://github.com/openfans-community-offical/Debian-Pi-Aarch64/blob/master/README.md#3-164-bluetooth-audio-connection-lost
In this project they got the same problem；
they used this way to fix it：
“3-16.4 Bluetooth Audio connection lost
Reason:

Pi's Bluetooth has compatibility issues with certain hardware between alsa and pulseaudio.

Solution:

Disable Bluealsa Service, just use pulseaudio for bluetooth audio. Run the command below and reboot:

systemctl mask  bluealsa.service
Note： Here U must use "mask" for unactive bluealsa.service, mustn't use "disable" !!”

Can we do in the same way？

NonstickAtom785, 2021-05-27:
I'm going to put some stress on this. It doesn't appear that the latest version of Chromium has any issues with bluetooth, but for some reason we do. What could be the issue?

fydeos-alpha, 2021-07-08:
Chromium OS does not have the best Bluetooth support, this is an inconvenient fact. For the original issue about "choppy", we could also reproduce this while high-res youtube video playback. Reducing the resolution or just playing a static audio file on a local drive sounds better. This is unfortunately due to the performance constrain of the Pi. 

We will look into continuously optimising Bluetooth performance but I sense this is a long shot. For immediate resolution, it's best to connect audio via HDMI or a 3.5mm jack.

ClowdyAllDay, 2021-08-05:
I play netflix and youtube on manjaro arm with bluetooth sound bar without choppy playback so i disagree that it is the constraints of the raspberry pi specifically.  perhaps a newer version of chromium os is needed?  If there is one.

fydeos-alpha, 2021-11-09:
fixed, try r92-r2 release

closed 2021-11-09 by fydeos-alpha