FydeOS/chromium_os-raspberry_pi #151: [Issue report] Enable debug serial port on FydeOS

ColinWang1203, 2021-03-14:
**Describe the bug or issue**
Hello, Is there a way to enable the debug boot log like RPI-OS?
On RPI-OS we can use USB to TTL and connect the uart to get the boot log.
Is there a similar way we can achieve this on the FydeOS?
I tried to add the "USE=pcserial" when building the packages and images but it wont work.
Really like this project and want to dig futher into this kernel, will be greatly appreciated if any help!

**Describe your setup**
 - 4B with 4GB RAM
 - Chromium OS r86
 - USB to TTL connected on the Gnd gpio14 and gpio15 on rpi

**To Reproduce**
flash image and boot, no log on the /dev/ttyUSB0

**Expected behavior**
log is visible via putty serial port

**Screenshots and/or hardware setup photos**

[image]

**Additional context**
Thanks!

ColinWang1203, 2021-03-16:
Any support? Really need to run this as an important project.

yunshansimon, 2021-03-17:
You need to modify the config.txt on EFI partition (the 12th partition on disk). 
```
mkdir /tmp/sdb12
sudo mount /dev/sdb12 /tmp/sdb12
sudo sh -c "echo   'enable_uart=1' >> /tmp/sdb12/config.txt"
sudo umount /dev/sdb12
```
/dev/sdb12 as the 12th of your usb disk.

Or you can modify the config.txt file at `overlay-rpi4/chromeos-base/chromeos-bsp-rpi4/files/kernel-config` , and remerge the package by ` emerge-rpi4 chromeos-bsp-rpi4` . Then rebuild the image. You will get an "uart enabled" image.

ColinWang1203, 2021-03-19:
> You need to modify the config.txt on EFI partition (the 12th partition on disk).
>
> ```
> mkdir /tmp/sdb12
> sudo mount /dev/sdb12 /tmp/sdb12
> sudo sh -c "echo   'enable_uart=1' >> /tmp/sdb12/config.txt" [...]

Hello yunshansimon,

Thanks for your reply,

I tried to uncomment the enable_uart=1 in the config.txt and rebuild&flash the image,
but kernel panic happens and I could not solve it for these days,
Would you mind look into it?

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2021.03.19 20:37:51 =~=~=~=~=~=~=~=~=~=~=~=
10fd083]
[    0.000000] Linux version 5.4.74-v8-g7341321b8bbb (colin@colin-UX410UQK) (Chromium OS 11.0_pre394483_p20200618-r9 clang version 11.0.0 (/var/cache/chromeos-cache/distfiles/egit-src/llvm-project b726d071b4aa46004228fc38ee5bfd167f999bfe)) #3 SMP PREEMPT Fri Mar 19 08:27:27 CST 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] percpu: Embedded 30 pages/cpu s85400 r8192 d29288 u122880
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1  smsc95xx.macaddr=DC:A6:32:9F:6A:EA vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  coherent_pool=6M smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 console=ttyS0,115200 earlyprintk=ttyS0,115200 log_level=7 root=PARTUUID=45960909-72D1-D84F-BBCD-4642FD907AD0 rootfstype=ext2 rootwait ro cros_debug cros_secure elevator=deadline
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 3627096K/4050944K available (9468K kernel code, 894K rwdata, 3000K rodata, 960K init, 929K bss, 161704K reserved, 262144K cma-reserved)
[    0.000000] random: get_random_u64 called from __kmem_cache_create+0xb4/0x598 with crng_init=0
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31291 entries in 123 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000165] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000186] pid_max: default: 32768 minimum: 301
[    0.000367] LSM: Security Framework initializing
[    0.000396] Yama: becoming mindful.
[    0.000430] LoadPin: ready to pin (currently not enforcing)
[    0.000648] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000697] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.002850] ASID allocator initialised with 32768 entries
[    0.003042] rcu: Hierarchical SRCU implementation.
[    0.003621] EFI services will not be available.
[    0.004130] smp: Bringing up secondary CPUs ...
[    0.004969] Detected PIPT I-cache on CPU1
[    0.005025] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.005999] Detected PIPT I-cache on CPU2
[    0.006036] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.006957] Detected PIPT I-cache on CPU3
[    0.006995] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.007138] smp: Brought up 1 node, 4 CPUs
[    0.007173] SMP: Total of 4 processors activated.
[    0.007188] CPU features: detected: 32-bit EL0 Support
[    0.007202] CPU features: detected: CRC32 instructions
[    0.007288] CPU: All CPU(s) started at EL2
[    0.007336] alternatives: patching kernel code
[... 178 lines cut ...]
[    1.993428] Freeing unused kernel memory: 960K
[    1.998010] Run /sbin/init as init process
/sbin/in[    2.037338] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00
it: erro[    2.041461] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
[    2.045739] CPU: 3 PID: 1 Comm: init Not tainted 5.4.74-v8-g7341321b8bbb #3
r while [    2.045741] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[    2.045743] Call trace:
loading [    2.045755]  dump_backtrace+0x0/0x144
[    2.045760]  show_stack+0x20/0x2c
[    2.054747] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    2.061790]  dump_stack+0xbc/0x150
[... cut ...]
[    2.061802]  __arm64_sys_exit_group+0x20/0x30
[    2.068415] usb 1-1: Product: USB2.0 Hub
[    2.070876]  el0_svc_common+0xac/0x198
[    2.070880]  el0_svc_compat_handler+0x2c/0x40
[    2.070882]  el0_svc_compat+0x8/0x10
[    2.070889] SMP: stopping secondary CPUs
[    2.076389] hub 1-1:1.0: USB hub found
[    2.078642] Kernel Offset: disabled
[    2.078645] CPU features: 0x0002,20006000
[    2.078647] Memory Limit: none
[    2.138429] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00 ]---

closed 2021-03-19 by ColinWang1203