FydeOS/chromium_os-raspberry_pi #154: Not able to turn off display

illivanillie, 2021-03-21:
**Describe the bug or issue**
Can’t get the display to sleep.
I have had installed keep awake. There it did not work (had the mode set to sun set). After Deinstallation it also did not work. 
The chrome OS settings are set to deactivate display during charging.
I added hdmi_blanking=1 to the confit file.

**Describe your setup**
 - Pi4, 8GB
 - Single monitor full hd Samsung
 - Newest release installed

**To Reproduce**
Steps to reproduce the behavior:
See above.

**Expected behaviour**
Expected the monitor to switch off/standby mode after a few minutes (8 minutes as I see it)

**Screenshots and/or hardware setup photos**

**Additional context**
I am not 100% sure about this (because it takes long to wait after every change and it was a few weeks ago): I think before I installed keep awake, the display went into standby mode after logging out of chrome os, but then it was not able to get out of the standby mode with mouse/keyboard activity.

yunshansimon, 2021-03-25:
We prevent the system to sleep for some kernel issue. It's at the bottom of the issues list.

illivanillie, 2021-03-25:
Maybe there’s a misunderstanding.
I am just talking about the monitor! Not the system. I don’t care if the system stays on. But usually (on any pc I’ve known before) it was possible to get the monitor into standby mode. That is what I wanted to achieve. The light is disturbing when the monitor stays on the whole day in my use case

fydeos-alpha, 2021-04-16:
I would suggest turning off your monitor manually by pressing a physical button, there must be one of those on your Samsung monitor 😂 Keep awake extension was tested to work with "sunny" mode to keep the system insomnia.

closed 2021-04-16 by fydeos-alpha