FydeOS/chromium_os-raspberry_pi #157: KIOSK mode - extracted or clone

ifzl, 2021-04-18:
To arrange the KIOSK mode, how do we need to do the following step in the process;

Place your desired extracted "Chrome app" to be launched under kiosk mode at /usr/local/share/kiosk_app/. For simplicity sake, you can clone the entire repository of our demo app and place it at /usr/local/share/kiosk_app/kiosk-demo-app

So how do we place the desired extracted or clone the entire repository ?

fydeos-alpha, 2021-04-18:
the easiest way would be:

1. Download the [Kiosk Demo App](https://github.com/FydeOS/kiosk-demo-app) as a zip file onto your Chromium OS download folder
2. Extract the zip file and obtain kiosk-demo-app folder in your Downloads folder
3. Use shell command `cp -r` or `mv` to copy/move the kiosk-demo-app folder to `/usr/local/share/kiosk_app/`

ifzl, 2021-04-20:
Thank you for you input, so how does this work with other App's ?

fydeos-alpha, 2021-05-13:
@ifzl  replace [kiosk demo app](https://github.com/FydeOS/kiosk-demo-app) with other apps.

closed 2021-05-13 by fydeos-alpha