FydeOS/chromium_os-raspberry_pi #158: Linux (Beta) can not be installed in the new Chromium r89r1

KarlJoachim, 2021-04-23:
**Describe the bug or issue**
A clear and concise description of what the bug is.

**Describe your setup**
 - Your Raspberry Pi model and spec variant, e.g. 4B with 2GB RAM
 - Your Chromium OS for Raspberry Pi release version, e.g. Chromium OS r77 r3
 - Your hardware configuration and setup, e.g. how you have wired the Pi board with any other peripheral connected to it

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots and/or hardware setup photos**
If applicable, add screenshots to help explain your problem. If your problem is hardware related, it's much easier to include a photo of how your setup looks like.

**Additional context**
Add any other context about the problem here.

mdali7053, 2021-05-09:
My raspberry pi 4b 2Gb install linux but error why

KarlJoachim, 2021-05-09:
I have repeated the installation of Chromium r89r1 and the Linux (beta) cannot be installed. It ends with an error. I use an Raspberry 4B with 8 GB RAM an a 32 GB sd-Card. I have made same pictures showing the disk space an the other windows during the installation of the Linux environement.
[image]
[image]
[image]
[image]

yunshansimon, 2021-05-11:
The Linux environment needs 4G memory/ 4G free hard disk space at least. The chromium OS for rpi needs downloading 1G binary from google's server, pls check your internet connection.

thepolarbear0818, 2021-05-15:
I have the same issue, with an 8G ram RPI and a 16G SD card, I have internet, but it will always fail on creating a disk. I tried this many times by re-writing the image to the sd card. It will not word. It did work on the last version though, so it's probable only this version that's affected.

NonstickAtom785, 2021-05-27:
This actually has nothing to do with the build itself. This is an issue with chromium os itself. Just wait until the next full dev build and it should be fixed.

KarlJoachim, 2021-05-28:
Ok. Thank you. The Linux environment in the r86r2 version is a very good feature, I like it.

tkchakos, 2021-07-07:
I am also having this. I have 8 GB of RAM & 425 GB of free disk space. I will wait for the next update. Thank you!

sindelar-fr, 2021-07-22:
Is there some plan on when the next release that will fix this will be? (October-ish?)

fydeos-alpha, 2021-07-22:
@sindelar-fr  working on that right now, stay tuned

fydeos-alpha, 2021-08-09:
Unfortunately we don't have plan to fix this for r89. It'll be fixed in r92 which is our next major release. For the time being, you guys can try FydeOS for You - Raspberry Pi 400 which has a working Crostini.

closed 2021-08-09 by fydeos-alpha