FydeOS/chromium_os-raspberry_pi #163: “Something went wrong” screen after trying to enable debugging options in the first time setup.

Temmie72832l, 2021-05-12:
**Describe the bug or issue**
“Something went wrong” screen after trying to enable debugging options in the first time setup

**Describe your setup**
 - Pi 4B 8GB RAM
 - Chromium 89.0.4389.82
 - USB 3.0 FD 32GB *booted from*
 -

**To Reproduce**
Steps to reproduce the behavior:
1. Boot into the first time setup
2. Click on 'Enable debugging features.'
3. See error

**Screenshots and/or hardware setup photos**
[image]
[image]

nirocfz, 2021-05-13:
There're a couple of reasons that can cause errors like this. Since the error occurred just after clicking the button, it might because of debugd(debug daemon process). 

I reproduced the error, and there's an error message in `/var/log/chrome/chrome` 

```
2021-05-13T06:01:51.779097Z ERROR chrome[1015:1015]: [object_proxy.cc(621)] Failed to call method: org.chromium.debugd.QueryDevFeatures: object_path= /org/chromium/debugd: org.chromium.debugd.error.DevFeatures: Unable to open FDT property firmware-version
```

It seems that this error is related to the firmware of the device.

The [image here](https://github.com/FydeOS/chromium_os-raspberry_pi/releases/tag/r89)  is a `dev` type image file, you can assume that the developer mode is already on, you can modify OS files, you can set a password for `chronos` or `root`, and ssh into the device.

Temmie72832l, 2021-05-13:
Thanks i'll be sure to try that out.

Temmie72832l, 2021-05-13:
Also what is the name of the user folder after sign in.

Temmie72832l, 2021-05-14:
Still experiencing the same issue with the img you linked

nirocfz, 2021-05-14:
I think you can ignore the error, you already got `debugging features`.

Temmie72832l, 2021-05-14:
just to clarify debugging features allow you to access / in the file manager correct?

nirocfz, 2021-05-14:
Please check the Introduction part of this page https://www.chromium.org/chromium-os/how-tos-and-troubleshooting/debugging-features

closed 2021-05-17 by nirocfz