FydeOS/chromium_os-raspberry_pi #164: Black screen with arrow after logo

achavesherra, 2021-05-13:
Hi!.
I have tried the last release in a Raspberry Pi 3b+ and I get a black screen after the Chromium logo on boot.
I can move the mouse arrow without problem but I never got to the login screen.

12dollar, 2021-05-18:
I've also tried the latest version, and provides me with the same issue. However, I only see the Chromium boot logo and a black screen thereafter.
Tried 4 different RPI, from 3b+ to 4 and different SDHC cards.

thiagoiwanko, 2021-05-27:
The last version also presented me with the same problem in a rasp 3b .. I had to use the chromiumos_test_image_r83r1-rpi3b.img.xz file version to make it work...

congratulations for those involved .. are doing a great job

mbah-dukun, 2021-06-01:
same problem with raspi 3b+

csjoy, 2021-07-21:
I'm having the same issue on my raspberry pi 3b+. After the boot logo only a black screen with the ability to move the mouse.

hurrellit, 2021-09-26:
Tried latest (r92) and previous (r89) on a Pi 3B+ with image burnt to standard microSD card and get the same: ChromiumOS splash screen is displayed on boot then just a black screen. Green activity indicator light is initially flashing but then activity just stops. Thanks for the work.

ghostyroasty, 2021-10-05:
Same issue on Pi 3B+. I was planning on using the OS in an elementary school computer lab to replace Kano OS.

AlbertPeronne, 2021-10-06:
Same here on Pi 3B+ with r92r1on microSD card (ethernet connected): Chromium splash screen, immediately followed by a black screen, and - if waiting long enough - a mouse cursor. Moving the cursor to the bottom left corner and clicking there shuts down the Pi...?

AlbertPeronne, 2021-11-01:
My issues are solved using the new dedicated release for Raspberry Pi 3 Model B and Model B+.
Thx !!!!!

closed 2021-11-02 by fydeos-alpha