FydeOS/chromium_os-raspberry_pi #167: Need help on remote access

giriannamalai, 2021-06-10:
**Describe the bug or issue**
This is not bug. It is wiki. I need to control the chromium rasperry from remote machine. The google remote desktop is changing the pin frequently. Its not reliable.
Any way to control it?

I have Pi4 with 4GB RAM

fydeos-alpha, 2021-06-18:
If by "remote access" you mean controlling the desktop remotely, there is no immediate solution to achieve your goal at the moment. We have implemented [FydeOS remote desktop](https://rdp.fydeos.com) that mimics Google's implementation, but it also needs user input for approval at the client-side. 

We have integrated our implementations into the enterprise offerings for FydeOS, admins from the management side are able to initiate RDP sessions and control managed (enrolled) clients. However this is a commercial offering.

closed 2021-06-18 by fydeos-alpha

samlark, 2023-03-27:
FydeOS-for-you also offers the RDP option - not only the Enterprise offering. 
So, if it's only for Enterprise, then consequently, you need to eliminate it from other offerings - namely from FydeOS-for-you.
And if it's not, then it certainly should work flawlessly... as it doesn't. Eg. if I start https://rdp.fydeos.com/ on my Android-tablet, I'm not able to log-in using my local account - it insists on a FydeOS account. So, no support possible.

fydeos-alpha, 2023-03-28:
rdp.fydeos.com is offered across all FydeOS releases; you need a FydeOS
account to use it. If you do not have a FydeOS account, you can sign up for
one for free. I am not sure what's the problem here.