FydeOS/chromium_os-raspberry_pi #168: Exit from Kiosk

giriannamalai, 2021-06-12:
Can not exit from Kiosk mode. After pressing ctrl + alt + s , the screen goes blank. Nothing happens

kumaravel2312, 2021-06-17:
Hi, Were you able to find the solution for this. I am stuck on this issue as well. Thank you.

giriannamalai, 2021-06-17:
> Hi, Were you able to find the solution for this. I am stuck on this issue as well. Thank you.

Not yet. Seems, there were no solution. Also, the kiosk mode reset on every restart. It will only usable for static web sites.

fydeos-alpha, 2021-07-08:
This is not an issue, it's just an unimplemented feature.

"Kiosk mode" is part of the FydeOS enterprise solution offering. In this project, we are shipping a "proof of concept" and aim to demonstrate how this works in code level. If you wish to add functionalities to fit your needs, at the moment you'll just have to get your hands dirty I am afraid.

closed 2021-07-08 by fydeos-alpha