FydeOS/chromium_os-raspberry_pi #170: Constant screen flashing while watching YouTube

vying1, 2021-06-14:
**Describe the bug or issue**
The screen will blank and then come back and blank again after a few seconds.  Videos are forced to h264 by extension.  Occurs with hardware video decode on and off.

**Describe your setup**
Pi 400, 4gb, HDMI to monitor, FydeOS 89

**To Reproduce**
Steps to reproduce the behavior:
1. Any video on YouTube

**Expected behaviour**
No flashing.

vying1, 2021-06-15:
Not sure if related since it still flashes with the video decode off, but video decode on has this error in chrome://gpu.

2021-06-15T03:00:49.192174Z ERROR chrome[1216:1216]: [v4l2_video_decode_accelerator.cc(508)] : could not create EGLImageKHR, index=5 texture_id=11582021-06-15T03:01:11.135394Z ERROR chrome[1216:1216]: [gl_surface_egl.cc(774)] : EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture2021-06-15T03:01:11.135629Z ERROR chrome[1216:1216]: [v4l2_video_decode_accelerator.cc(508)] : could not create EGLImageKHR, index=0 texture_id=16362021-06-15T03:01:11.136474Z ERROR chrome[1216:1216]: [gl_surface_egl.cc(774)] : EGL Driver message (Critical) eglCreateImageKHR: dri2_create_image_khr_texture2021-06-15T03:01:11.136667Z ERROR chrome[1216:1216]: [v4l2_video_decode_accelerator.cc(508)] : could not create EGLImageKHR, index=1 texture_id=1637

vying1, 2021-06-16:
Problem is solved by using fkms instead of kms in config.txt.  But then there is no vsync...

yunshansimon, 2021-09-06:
Use `fkms` instead. Or you need insert flag `--fydeos-force-cursor-compositing` into `/etc/chrome_dev.conf`

vying1, 2021-09-13:
This didn't work for me.  Here are the contents of /etc/chrome_def.conf.  The screen still keeps flashing.

--load-media-router-component-extension=1
--fydemina
--disable-arc-opt-in-verification
--arc-start-mode=always-start-with-no-play-store
--enable-hardware-overlays=single-fullscreen
--fydeos-force-show-cursor
--enable-features=UseAlternateVideoDecoderImplementation
--use-vulkan=swiftshader
--enable-gpu-rasterization
--fydeos-force-cursor-compositing

vying1, 2021-09-13:
Ok, that flag didn't work on the live image but it does after installing.  The cursor is very sluggish now though.  Oh well.

fydeos-alpha, 2021-12-10:
fixed in r2 release.

closed 2021-12-10 by fydeos-alpha