FydeOS/chromium_os-raspberry_pi #171: Bluetooth mouse and keyboard connected but not responding

Qave55, 2021-06-22:
Bluetooth mouse and keyboard connected but not responding
MS Designer Mouse and Keyboard

Describe your setup

Your Raspberry Pi model 4B with 4GB RAM
Your Chromium OS for Raspberry Pi release version, e.g. Chromium OS version 89.0.4389.82 
Your hardware configuration and setup, e.g. how you have wired the Pi board with any other peripheral connected to it - None

To Reproduce Steps to reproduce the behavior:

Settings
Bluetooth
mouse found
select with USB mouse
pair
connected with battery life %
no mouse pointer movement

Settings
Bluetooth
Keyboard found
select with USB mouse
pair
enter pairing code on keyboard
connected with battery life %
no keyboard typing appears on screen

Expected behaviour mouse movement should move pointer on screen
Expected behaviour keyboard typing on screen 

connection times out possibly due to no use

jbkcrash, 2021-07-23:
I am noticing similar issue on pi400 I just setup with latest image. Using a Microsoft Mouse, it shows connected in the bluetooth manager. I get the event indicating it was succesfully added, but it never actually moves the mouse cursor. I haved removed and tried pairing multiple times with same result.

I would be able to do some debugging if there is anything you want me to look at.

fydeos-alpha, 2021-12-10:
bluetooth issue fixed in r2 release.

closed 2021-12-10 by fydeos-alpha