FydeOS/chromium_os-raspberry_pi #184: chrome://flags Page is blank/empty on r92 release

carlosg2, 2021-08-25:
Unable to use chrome://flags page on r92 release.
Displays blank/empty list, only the search field and reset button are shown.

**Setup**
Raspberry Pi 4B with 8GB RAM
Chromium OS for Raspberry Pi r92 r1

**To Reproduce**
1. Open Chromium browser
2. Navigate to 'chrome://flags'
3. Flags page is displayed with an empty list, only the search field and reset button are shown
(same result under incognito mode)

**Expected behaviour**
Display a list of flags available for Chromium OS/browser.  

[image]

**Additional info**
A quick google search for 'chrome flags empty' shows problem with chrome os in the past related to TPM (Trusted Platform Module) but i cannot confirm.

5erade, 2021-09-03:
Exact same problem...

fydeos-alpha, 2021-11-09:
fixed, try r92-r2 release

closed 2021-11-09 by fydeos-alpha