FydeOS/chromium_os-raspberry_pi #186: [Issue report] Linux Does not work on raspberry pi 4B with 4gb ram.

ABUCKY0, 2021-08-28:
**Describe the bug or issue**
A clear and concise description of what the bug is.

**Describe your setup**
 - Your Raspberry Pi model and spec variant, e.g. 4B with 2GB RAM
 - Your Chromium OS for Raspberry Pi release version, e.g. Chromium OS r77 r3
 - Your hardware configuration and setup, e.g. how you have wired the Pi board with any other peripheral connected to it

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings'
2. Click on 'LDE( Linux Development Enviroment'
3. Scroll down to 'Turn On'
4. See error

**Expected behaviour**
A clear and concise description of what you expected to happen.
LDE to turn on without a hitch.
**Screenshots and/or hardware setup photos**
If applicable, add screenshots to help explain your problem. If your problem is hardware related, it's much easier to include a photo of how your setup looks like.

**Additional context**
Add any other context about the problem here.

closed 2021-08-28 by fydeos-alpha

ABUCKY0, 2021-08-28:
What is funny is the fact that the issue was closed with zero input on how to fix it.

fydeos-alpha, 2021-08-29:
Not really funny, confused rather - confused why your issue was closed without reason?

Whoever had the intention to help you would be confused too. This is not how you raise an issue. The issue template was supposedly for you to use as a "template", not keeping the original content and post it as the "issue".

Issues like yours would be treated as spam.

Learn how to describe your issue and provide useful information before posting.

ABUCKY0, 2021-08-29:
> Not really funny, confused rather - confused why your issue was closed without reason?
>
> Whoever had the intention to help you would be confused too. This is not how you raise an issue. The issue [...]

Steps to learn something:

1. Figure out what you want to learn.
2. Find someone or someone’s paper/article/instructions to teach you.
3. Success.

If there is no instructions on the topic, there is no way to learn.