FydeOS/chromium_os-raspberry_pi #195: Chromiumos r92-r2 does not boot

KarlJoachim, 2021-11-11:
I have a Raspberry 4B with 4 GB RAM and use a SD-Card with 32 GB. I downloaded the image file, unzip the file and used the Raspberry image tool for erasing the SD-Card. Afterwards I wrote the image file with the raspberry tool to the SD-Card. This is the same way as I preceded with the r92 and r86-r2 which booted properly.

fydeos-alpha, 2021-11-11:
how does it not boot? black screen after rainbow or what?

sindelar-fr, 2021-11-11:
This is also happening to me. First it didn't want to recognize the sdcard at all - it went to the bootloader where I could see it trying to boot.(I just wrote the img file using dd to sdcard as usual). I ran gparted, it complained about the free space being reported wrong, fixed it, popped the card back to the raspi and now the screen briefly notices there's something plugged into the hdmi port and then goes to sleep saying no hdmi connected. Green LED seems to flash time to time. I currently don't have means to try to connect to it over the network, maybe later in the evening.

ryotaw, 2021-11-12:
It's happen to me too. No rainbow. HDMI signal came to display and a few seconds later stop. Repeat that a several time and finally completely no signal.
I am using a WQHD(2560 * 1440) size display.
r92-r1 runs fine on the same environment.

yunshansimon, 2021-11-12:
This version graphic driver is changed to 'kms' for better performance. But it may need more power than 'fkms'. Some solutions:(all of them need editing the config.txt in partition 12 of image)
Try to keep with "kms" :
1. Uncomment the line `config_hdmi_boost=7`. 
2. Unplug all of your devices except the monitor to reduce the power consumption. Plug them after booting. If it works, you need a lower power-consuming keyboard or mouse, etc.
3. Change your boot device from USB to SD card to reduce the power consumption.
If all of them don't work, you have to go back to kms:
1. Uncomment the line `dtoverlay=vc4-fkms-v3d` , comment the line `dtoverlay=vc4-kms-v3d`

If you rpi4 only got 1G ram, it may not boot:
1. Change the line `gpu_mem=256` to `gpu_mem=128`, to keep the hardware decoder running with low resolution.

Total go back:
1. Comment `gpu_mem=256`
2. Uncomment 'dtoverlay=vc4-fkms-v3d'
3. Comment `dtoverlay=vc4-kms-v3d`
4. Comment `--fydeos-force-show-cursor --enable-rpi-video-decoder --no-sandbox` in `chrome_dev.conf` at `/etc` in the partition 3 of image.

KarlJoachim, 2021-11-15:
Thank you yunshansimon.

I made your three suggestions in the config.sys:
 - uncomment  "config_hdmi_boost=7"
 - uncomment  "dtoverlay=vc4-fkms-v3d"
 - comment      "dtoverlay=vc4-kms-v3d"

and now chromium ist booting (visible).

closed 2021-11-16 by fydeos-alpha