FydeOS/chromium_os-raspberry_pi #200: Double écran impossible sur RPI4

Stee29, 2021-12-10:
Bonjour

le 2ème écran ne s'affiche pas et reste noir, alors qu'il est bien détecté (en HDMI ou en VGA)
la souris disparait sur le 2e écran et n'est pas bloquée par la fin du cadre, elle passe bien à coté

fydeos-alpha, 2021-12-10:
in English, please

Stee29, 2021-12-10:
Hello

the 2nd screen is not displayed and remains black, although it is correctly detected (in HDMI or VGA)
the mouse disappears on the 2nd screen and is not blocked by the end of the frame, it goes right next to it

fydeos-alpha, 2021-12-10:
Unfortunately, we have not tested dual-monitor support with this release, yet. Consider it unsupported for now, we will take a look and may include this for the coming releases

closed 2021-12-10 by fydeos-alpha

Stee29, 2021-12-10:
perfect thank you