FydeOS/chromium_os-raspberry_pi #213: TP-Link Archer T3U AC1300 USB WiFi adaptor

elpraga, 2022-02-20:
**Describe the bug or issue**
I have bought the **TP-Link Archer T3U AC1300 USB WiFi adaptor**, it significantly boosts my WiFi signal. It does not work out of the box in  neither **Chromium OS** (r92-r2) nor **Ubuntu 20.04**. Nonetheless I was able to make it work under Ubuntu following the instructions in https://community.tp-link.com/en/home/forum/topic/208022?page=1 .

First question first. Can USB WiFi adaptors be used with Chromium OS in general? 

**Describe your setup**
 - Your Raspberry Pi model and spec variant, e.g. 4B with 4GB RAM
 - Your Chromium OS for Raspberry Pi release version, e.g. Chromium OS r92 r2
 - Your hardware configuration and setup, e.g. how you have wired the Pi board with any other peripheral connected to it - no, just external bluetooth keyboard and a wireless mouse. Both work fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug the TP-Link Archer T3U AC1300 USB WiFi adaptor in
2. it cannot be used.

**Expected behaviour**
It can be used to boost the WiFi speed. 

**Additional context**
The process of getting it work under Ubuntu involved compiling the driver and setting up dkms.

```
git clone https://github.com/cilynx/rtl88x2bu.git
cd rtl88x2bu
VER=$(sed -n 's/\PACKAGE_VERSION="\(.*\)"/\1/p' dkms.conf)
sudo rsync -rvhP ./ /usr/src/rtl88x2bu-${VER}
sudo dkms add -m rtl88x2bu -v ${VER}
sudo dkms build -m rtl88x2bu -v ${VER}
sudo dkms install -m rtl88x2bu -v ${VER}
sudo modprobe 88x2bu
```

1. Can this driver be included in future Chromium OS releases?
2. Can it be compiled within the **linux environment** under Chromium OS and then be used within the GUI?

elpraga, 2022-02-22:
Well, I have made (a very limited) progress. 

1. I have found out that the kernel version available in the linux subsystem (4.19) does not match the kernel version used by Chromium OS (5.4.83 according to `uname -a`).
2. I have also found the kernel you use (https://github.com/FydeOS/kernel-raspberry_pi) -- which seems not too big to be downloaded and unzipped (216 MB of the zip file) . I probably do not need all of it, but I am not experienced enough to know which parts exactly I need without trying it first.
3. The `Makefile` in the rtl88x2bu driver has several options to use for arm devices (even one directly for RPi, even though I am not sure if this should be used in this case) -- Could you guide me which option to use, and I'll try to compile the driver when I have some time on my hands? 
[Makefile.txt](https://github.com/FydeOS/chromium_os-raspberry_pi/files/8113453/Makefile.txt)

elpraga, 2022-02-22:
These are the platform related options in the `Makefile` :
```
###################### Platform Related #######################
CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_ANDROID_X86 = n
CONFIG_PLATFORM_ANDROID_INTEL_X86 = n
CONFIG_PLATFORM_JB_X86 = n
CONFIG_PLATFORM_ARM_S3C2K4 = n
CONFIG_PLATFORM_ARM_PXA2XX = n
CONFIG_PLATFORM_ARM_S3C6K4 = n
CONFIG_PLATFORM_MIPS_RMI = n
CONFIG_PLATFORM_RTD2880B = n
CONFIG_PLATFORM_ARM_NV_NANO = n
CONFIG_PLATFORM_MIPS_AR9132 = n
CONFIG_PLATFORM_RTK_DMP = n
CONFIG_PLATFORM_MIPS_PLM = n
CONFIG_PLATFORM_MSTAR389 = n
CONFIG_PLATFORM_MT53XX = n
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM_MX51_241H = n
CONFIG_PLATFORM_FS_MX61 = n
CONFIG_PLATFORM_ACTIONS_ATJ227X = n
CONFIG_PLATFORM_TEGRA3_CARDHU = n
CONFIG_PLATFORM_TEGRA4_DALMORE = n
CONFIG_PLATFORM_ARM_TCC8900 = n
CONFIG_PLATFORM_ARM_TCC8920 = n
CONFIG_PLATFORM_ARM_TCC8920_JB42 = n
CONFIG_PLATFORM_ARM_TCC8930_JB42 = n
CONFIG_PLATFORM_ARM_RK2818 = n
CONFIG_PLATFORM_ARM_RK3066 = n
CONFIG_PLATFORM_ARM_RK3188 = n
CONFIG_PLATFORM_ARM_RK3399 = n
CONFIG_PLATFORM_ARM_URBETTER = n
CONFIG_PLATFORM_ARM_TI_PANDA = n
CONFIG_PLATFORM_MIPS_JZ4760 = n
CONFIG_PLATFORM_DMP_PHILIPS = n
CONFIG_PLATFORM_MSTAR_TITANIA12 = n
CONFIG_PLATFORM_MSTAR = n
CONFIG_PLATFORM_SZEBOOK = n
CONFIG_PLATFORM_ARM_SUNxI = n
CONFIG_PLATFORM_ARM_SUN6I = n
CONFIG_PLATFORM_ARM_SUN7I = n
CONFIG_PLATFORM_ARM_SUN8I_W3P1 = n
CONFIG_PLATFORM_ARM_SUN8I_W5P1 = n
CONFIG_PLATFORM_ACTIONS_ATM702X = n
CONFIG_PLATFORM_ACTIONS_ATV5201 = n
CONFIG_PLATFORM_ACTIONS_ATM705X = n
CONFIG_PLATFORM_ARM_SUN50IW1P1 = n
CONFIG_PLATFORM_ARM_RTD299X = n
CONFIG_PLATFORM_ARM_LGE = n
CONFIG_PLATFORM_ARM_SPREADTRUM_6820 = n
CONFIG_PLATFORM_ARM_SPREADTRUM_8810 = n
CONFIG_PLATFORM_ARM_WMT = n
CONFIG_PLATFORM_TI_DM365 = n
CONFIG_PLATFORM_MOZART = n
CONFIG_PLATFORM_RTK119X = n
CONFIG_PLATFORM_RTK119X_AM = n
CONFIG_PLATFORM_RTK129X = n
CONFIG_PLATFORM_RTK390X = n
CONFIG_PLATFORM_NOVATEK_NT72668 = n
CONFIG_PLATFORM_HISILICON = n
CONFIG_PLATFORM_HISILICON_HI3798 = n
CONFIG_PLATFORM_NV_TK1 = n
CONFIG_PLATFORM_NV_TK1_UBUNTU = n
CONFIG_PLATFORM_RTL8197D = n
CONFIG_PLATFORM_AML_S905 = n
CONFIG_PLATFORM_ZTE_ZX296716 = n
```

fydeos-alpha, 2022-03-20:
Support for specific peripheral devices is out of scope for this project, per the FAQ. I am sorry, if you find anything else useful you are welcome to share it here but I will have to close this for now.

closed 2022-03-20 by fydeos-alpha

elpraga, 2022-03-22:
I do understand and thank you for your answer. 

Could you leave the issue open, just in case I it someone else can make some progress?

elpraga, 2022-03-22:
And before I even try to begin, is it technically possible to use an external USB wifi dongle with chromium os @fydeos-alpha ?

I have no way of reliably finding that out..