FydeOS/chromium_os-raspberry_pi #217: No Audio over HDMI in release r96

KarlJoachim, 2022-03-30:
I have noticed that the audio over HDMI does not work any more in the verson r96. I used/use r92r2 and there the audio over HDMI works pretty with exactly the same equipment. Did you made there a change?

I have a Raspi 4B with 4 GB a lenovo LEN L27q-30 with loudspeakers.

netbug1010, 2022-04-22:
I am on r96-r1. All the audio output device names are the same as `Speaker(Internal)`. The audio didn't work by default, it worked when I changed output to one of them.

KarlJoachim, 2022-04-25:
Where did you change the output? In the config.txt file?

netbug1010, 2022-04-27:
Click on the system tray at the right bottom of screen, click on the `>` at the right of the volume bar, you will see the audio devices, change to one of them.

KarlJoachim, 2022-04-27:
Thank you!! Problem solved.

closed 2022-04-27 by KarlJoachim