FydeOS/chromium_os-raspberry_pi #218: tlsdate patch request.

Alex313031, 2022-04-01:
Hello, I am using parts of your overlay in my own chromiumos builds. > https://github.com/Alex313031/ChromiumOS

I noticed in //chromium_os-raspberry_pi/overlay-rpi4-fydeos/virtual/chromeos-bsp/chromeos-bsp-0.0.1.ebuild
there is a reference to net-misc/patch-tlsdate which I am assuming is to fix the bug where ChromiumOS will not fetch the timezone or time automatically.
I do not see this in the ChromiumOS source, or an ebuild for it here or in your chromiumos-vmware repo, so I am assuming it is internal.
Would you mind sharing this ebuild or what I need to modify in the default net-misc/tlsdate ebuild/package to fix this?

fydeos-alpha, 2022-04-02:
The `overlay-rpi4-fydeos` overlay is for FydeOS for Raspberry Pi 400, not Chromium OS for Raspberry Pi. 

As stated in the FAQ:
> This project is about Chromium OS for Raspberry Pi, not FydeOS for You - Raspberry Pi 400. However, some of the elements that we have used to produce FydeOS for You - Raspberry Pi 400 are also open-sourced in this project.

If you wish to build Chromium OS, please use [overlay-rpi4](https://github.com/FydeOS/chromium_os-raspberry_pi/tree/master/overlay-rpi4).

On a further note about `patch-tlsdate`, it was not intended to fix Chromium OS not fetching the timezone, actually Chromium OS does automatically fetch the timezone (though it relies on a chargeable Google API). It was a patch to replace the defaut tls server that is more neutral and offers availability for users under censored internet accessibility.

Alex313031, 2022-04-03:
@fydeos-alpha Ahhh ok I see. I have API keys and build chromeos-chrome locally (thanks for the libwidevine patch too), but not ones that require payment. Do you know what API I need to enable to fix this?

Also, I'm continuing the work of arnoldthebat and so am only using bits and pieces from this overlay, as the ultimate goal of my project is to make an amd64 release that will have codecs, widevine, and google drive and be usable on as many devices as possible.

I've also used bits and pieces from your vmware overlay, nayuos's overlay, and cloudready's publicly available ebuilds.

fydeos-alpha, 2022-04-03:
This conversation is outside the scope of this project so I am going to have to close it.

iirc Chromium OS requires the [geolocation API](https://developers.google.com/maps/documentation/geolocation/overview) and the [timezone API](https://developers.google.com/maps/documentation/timezone/overview) for your desired feature to work, I may be wrong, do check with the [Chromium-OS-Dev](https://groups.google.com/a/chromium.org/g/chromium-os-dev) guys if you want a more accurate answer.

closed 2022-04-03 by fydeos-alpha