FydeOS/chromium_os-raspberry_pi #220: Unable to enable the Linux Development Environment

netbug1010, 2022-04-22:
**Describe the bug or issue**
When enabling, returns "Error downloading the virtualmachine, Please try again." 

**My setup**
 - Raspberry Pi 4B with 8GB RAM
 - image r96, r96-r1
 - USB thumb drive, 64G

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'setting'
2. Click on 'Developer', click on "Turn on"
3. Using the default setting
4. Click "Install"

**Expected behaviour**
Linux development environment is installed/enabled.

**Additional context**
I used r92 image and everything went fine on the same hardware.

KarlJoachim, 2022-04-24:
Did yout expand the partition? See below the issue storage amount incorrect.

netbug1010, 2022-04-24:
Yes, I have 46+G available on state partition.

locness3, 2022-08-20:
Same issue, I have expanded the partition (if there's not enough free space, it displays an error before attemtping to download)

netbug1010, 2022-09-23:
Just try the r102-r2 image, it is working fine with sdcard.

closed 2022-10-29 by fydeos-alpha

locness3, 2022-11-02:
I'll have a look when possible