FydeOS/chromium_os-raspberry_pi #221: Chrome storage settings reports I only have 3.1 GB free

slannon97, 2022-04-25:
I flashed the latest image to a 32 GB card. However, Chromium OS storage settings reports I only have 3.1 GB available. Why is this? Is it a bug? I would have expected something closer to 21 GB.

pranavr2003, 2022-05-25:
@slannon97 Did you try expanding the partition? Try connecting the SSD/USB/SD card to another OS and expand the partition with GParted or anything else

slannon97, 2022-05-30:
Thank you for the information. Can this functionality be added to the image in the future so it does it automatically?

pranavr2003, 2022-05-31:
It can be made to run on first boot, probably. It's just a shell command. I'm not very good with writing shell scripts myself, so I'm not very sure where I would put it. If you are aware of this, maybe a PR would be good.

fydeos-alpha, 2022-10-29:
The stateful partition expansion it's already in the initialisation process, but for some reason it didn't work for you.

closed 2022-10-29 by fydeos-alpha