FydeOS/chromium_os-raspberry_pi #223: [Issue report] Tablet gestures not working with official Raspberry Pi Touch Display

cletusw, 2022-05-01:
**Describe the bug or issue**
[Tablet Mode gestures](https://support.google.com/chromebook/answer/9739838) (except swipe from left to go back) not working.

**Describe your setup**
 - Raspberry Pi 4B with 2GB RAM
 - Chromium OS r96 r1
 - Pi connected via DSI + 5V + GND to an [official Raspberry Pi Touch Display](https://www.raspberrypi.com/products/raspberry-pi-touch-display/)
    - This required switching from KMS to FKMS in config.txt: `dtoverlay=vc4-fkms-v3d`

**To Reproduce**
1. Enter Tablet Mode
    1. [Enable "Debugging keyboard shortcuts" flag](https://www.reddit.com/r/chromeos/comments/fxnd1u/comment/fz4z6rd/)
    2. Click the Restart button
    3. Press Ctrl + Alt + Shift+ T to switch to Tablet Mode
2. Open an app
3. Swipe up from the bottom -- tried long, short, swipe + hold

**Expected behaviour**
OS should show Home or Pinned apps or All open app windows, respectively.

**Actual behavior**
A regular scroll happens

Note swiping in from the left edge of the screen to go back works fine, so I don't think it's a hardware issue.

cletusw, 2022-05-01:
Also any pointers on booting into Tablet Mode would be great since I don't intend to have a physical keyboard connected once set up, thanks!

fydeos-alpha, 2022-05-01:
This is a known limitation to Chromium OS, as the Chromium OS we released believes you are having a "Chromebox", not a Chrometablet. At this moment we do not have any intention to change this behaviour and it is thus outside the scope of this project.

closed 2022-05-01 by fydeos-alpha