FydeOS/chromium_os-raspberry_pi #229: Extremely slow login

Turux, 2022-08-02:
After the initial boot and the creation of the user account, the login process is incredibly slow (up to 2 minutes to login)

**Describe your setup**
 - Raspberry Pi 4B 4GB, 128GB SD card 
 - r102
 - 30W power supply, USB mouse, USB keyboard, hdmi monitor

**To Reproduce**
Steps to reproduce the behavior:
1. Install Chromium OS r102
2. Create an account 
3. Reboot 
4. Attempt to login
5. See the error 

**Expected behaviour**
Up until r96, with same exact hardware, login took circa 5 seconds, now takes minutes

Turux, 2022-08-13:
Tested with r102-r1.
Problem still exists.

peeweep, 2022-08-24:
Hi @Turux, try r102-r2

Turux, 2022-08-26:
Hi @peeweep, I just tried with r102-r2 and indeed the problem is gone!
Thanks for your hard work! This is undoubtedly the best desktop experience with a rpi4

closed 2022-08-26 by Turux