FydeOS/chromium_os-raspberry_pi #231: Build with custom touchscreen driver

thomaswu0613, 2022-08-11:
Greetings, I have a raspberry pi MHS 3.5-inch touchscreen that uses GPIO. Is it possible to build the OS with custom overlay drivers? Also does FydeOS support GPIO? Thank you.

peeweep, 2022-08-11:
Of course you can build your system with a custom overlay driver.
There is a [config.txt](https://github.com/FydeOS/chromium_os-raspberry_pi/blob/r96-dev/overlay-rpi4/chromeos-base/chromeos-bsp-rpi4/files/kernel-config/config.txt) in our system, you can modify it before you build image, or use the `edit-pi-config` built into the OS to edit it.
And also FydeOS/Chromium OS support GPIO. In this system image we have built in some simple and useful tools, you can try `gpio*` commands. For example, `gpioinfo`.

thomaswu0613, 2022-08-11:
Since FydeOS is Linux based, can I use the driver overlay files and configs [here](https://github.com/FydeOS/chromium_os-raspberry_pi#setup-raspberry-pi-overlay)? And can you show me how to build the OS with the drivers above? Thanks !

peeweep, 2022-08-11:
A script of MHS35: https://github.com/goodtft/LCD-show/blob/master/MHS35-show
I think the part that can be used on Chromium OS in this script is to copy the dtb file and modify rpi config.txt.
I don't know how to drive your touchscreen display, need you to try it yourself.

If you want to have a specific reference, you can see https://github.com/fydeos/chromium_os-raspberry_pi/commit/1eae303e3
In order to let MIPI camera work, we modified rpi config.txt, camera lib, GBM lib, and kernel config in this commit.

peeweep, 2022-08-11:
> You are asking about issues about a 3rd-party app, a non-standard peripheral device or a special setup that does not benefit the general community

As FAQ, I will close this issue. If you find something useful , welcome to share with us.

closed 2022-08-11 by peeweep