FydeOS/chromium_os-raspberry_pi #235: Error while setting up network

sserve-kr, 2022-10-07:
When I try to connect to network, it returns "Error configuring network".
I tried to figure out what is the problem, but it says nothing more than that.

 - Raspberry Pi 4B (RAM 8GB)
 - Chromium OS r102 r2 rpi4b
 - Power (Type C cable), Micro HDMI connected with monitor, keyboard, mouse connected.
 
[image]

**To Reproduce**
1. Turn on raspberry pi
2. Connect keyboard, mouse
3. Try to connect to a network
4. See error

**Expected behaviour**
[image]
That's all of this problem.
It returns that red text and keeps fail to connect to network.

fydeos-alpha, 2022-10-29:
It seems to be wifi hotspot compatibility issues, does the wifi on your Pi work if you boot another OS?

sserve-kr, 2022-10-30:
> It seems to be wifi hotspot compatibility issues, does the wifi on your Pi work if you boot another OS?

yes. I installed Ubuntu 22.04 with same raspberry pi, and it connected to my wifi without any problem.

himehowareu, 2022-12-24:
I have the same problem , do you need anymore information on this issue , or is there any work arounds  as this make the rest of the OS unreachable.

psychicgoldfish, 2022-12-26:
same issue here.

jwaxy, 2023-08-29:
I'm having the same issue with my raspi4 is there any workaround or fixes? It became 2023

fydeos-alpha, 2023-08-30:
@jwaxy  well, it's worth revisiting. What's the detailed spec of your Pi 4? When and where was it manufactured? A photo of it would be very helpful.

jwaxy, 2023-08-30:
@fydeos-alpha Sure, it's a Raspberry Pi 4 Model B 2GB

[image]

maroonmoon13, 2023-11-20:
I have the same issue. I feel its a wifi country issue. I am using the pi in EU and cannot connect to wi-fi.

sonicwind16, 2023-12-02:
I'm having the exact same issue. I was able to connect to wifi by remvoing the password requirement/encryption... Def not recommended. Do we know what WPA formats Fydos supports?

dc-me, 2023-12-03:
Same here, wifi can only connect to open wifi.

fydeos-alpha, 2023-12-21:
The latest release (r114) should fix this issue. Sorry guys, it's been too long, I know 🫠

closed 2023-12-21 by fydeos-alpha