FydeOS/chromium_os-raspberry_pi #241: Failed to emerge chromeos-base/chrome-icu-96.0.4664.208_rc-r1 for /build/rpi4/, Log file:

wwd-615, 2023-03-30:
I get the following error when executing./build_packages --board=rpi4 --nowithautotest. Can you help me
>>> 11:21:16 Failed to emerge chromeos-base/chrome-icu-96.0.4664.208_rc-r1 for /build/rpi4/, Log file:
>>> 11:21:16   /build/rpi4/tmp/portage/logs/chromeos-base:chrome-icu-96.0.4664.208_rc-r1:20230329-032107.log
>>> 11:21:16 === Start output for job chrome-icu-96.0.4664.208_rc-r1 ===
chrome-icu-96.0.4664.208_rc-r1:  * Package:    chromeos-base/chrome-icu-96.0.4664.208_rc-r1
chrome-icu-96.0.4664.208_rc-r1:  * Repository: cros-pi
chrome-icu-96.0.4664.208_rc-r1:  * USE:        arm cups elibc_glibc kernel_linux libcxx neon runhooks userland_GNU xkbcommon
chrome-icu-96.0.4664.208_rc-r1:  * FEATURES:   network-sandbox splitdebug userpriv
chrome-icu-96.0.4664.208_rc-r1:  * Running stacked hooks for pre_pkg_setup
chrome-icu-96.0.4664.208_rc-r1:  *    sysroot_build_bin_dir ...
chrome-icu-96.0.4664.208_rc-r1:  [ ok ]
chrome-icu-96.0.4664.208_rc-r1:  * Using python2.7 to build
chrome-icu-96.0.4664.208_rc-r1:  * Running stacked hooks for post_pkg_setup
chrome-icu-96.0.4664.208_rc-r1:  *    python_eclass_hack ...
chrome-icu-96.0.4664.208_rc-r1:  [ ok ]
chrome-icu-96.0.4664.208_rc-r1: 11:21:13 >>> Unpacking source...
chrome-icu-96.0.4664.208_rc-r1:  * CHROME_ORIGIN VALUE is LOCAL_SOURCE
chrome-icu-96.0.4664.208_rc-r1:  * ERROR: chromeos-base/chrome-icu-96.0.4664.208_rc-r1::cros-pi failed (unpack phase):
chrome-icu-96.0.4664.208_rc-r1:  *   /mnt/host/source/fydeos/chromium does not contain a valid chromium checkout!
chrome-icu-96.0.4664.208_rc-r1:  * 
chrome-icu-96.0.4664.208_rc-r1:  * Call stack:
chrome-icu-96.0.4664.208_rc-r1:  *               ebuild.sh, line  125:  Called src_unpack
chrome-icu-96.0.4664.208_rc-r1:  *             environment, line 4455:  Called die
chrome-icu-96.0.4664.208_rc-r1:  * The specific snippet of code:
chrome-icu-96.0.4664.208_rc-r1:  *                   die "${CHROME_ROOT} does not contain a valid chromium checkout!";
chrome-icu-96.0.4664.208_rc-r1:  * 
chrome-icu-96.0.4664.208_rc-r1:  * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:chrome-icu-96.0.4664.208_rc-r1:20230329-032107.log
chrome-icu-96.0.4664.208_rc-r1:  * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/chrome-icu-96.0.4664.208_rc-r1/temp/build.log
chrome-icu-96.0.4664.208_rc-r1:  * CWD: /build/rpi4/tmp/portage/chromeos-base/chrome-icu-96.0.4664.208_rc-r1/work
chrome-icu-96.0.4664.208_rc-r1:  * S:   /build/rpi4/tmp/portage/chromeos-base/chrome-icu-96.0.4664.208_rc-r1/work/chrome-icu-96.0.4664.208_rc

ifttl, 2023-03-30:
It looks like you are using the r96-dev branch of the code. Please remove the CHROME_ROOT line from project-cros-pi/make.conf, and then re-run cros_sdk using --chrome-root to point to the directory of the Chromium source code and enter the chroot environment. You may also want to try using the updated r102-dev or r108-dev branches.

nirocfz, 2023-03-30:
It looks like you are using the r96-dev branch of the code.
Please remove the `CHROME_ROOT` line from project-cros-pi/make.conf, and then re-run `cros_sdk` using `--chrome-root` to point to the directory of the Chromium source code and enter the chroot environment.

You may also want to try using the updated r102-dev or r108-dev branches.

wwd-615, 2023-03-30:
thank you very much, let me try

wwd-615, 2023-03-31:
When I changed the make.conf file, I still got the following error，can you help me：

>>> 14:16:10 Failed to emerge chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1 for /build/rpi4/, Log file:
>>> 14:16:10   /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-96.0.4664.208_rc-r1:20230331-061607.log
>>> 14:16:10 === Start output for job chromeos-chrome-96.0.4664.208_rc-r1 ===
chromeos-chrome-96.0.4664.208_rc-r1:  * Package:    chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1
chromeos-chrome-96.0.4664.208_rc-r1:  * Repository: cros-pi
chromeos-chrome-96.0.4664.208_rc-r1:  * USE:        accessibility afdo_use arm autotest build_tests buildcheck cfi chrome_debug chrome_media chrome_remoting cros-debug debug_fission dwarf5 elibc_glibc fonts highdpi kernel_linux libcxx nacl neon oobe_config opengles orderfile_use ozone_platform_default_gbm ozone_platform_gbm ozone_platform_headless runhooks thinlto userland_GNU xkbcommon
chromeos-chrome-96.0.4664.208_rc-r1:  * FEATURES:   network-sandbox splitdebug userpriv
chromeos-chrome-96.0.4664.208_rc-r1:  * Running stacked hooks for pre_pkg_setup
chromeos-chrome-96.0.4664.208_rc-r1:  *    sysroot_build_bin_dir ...
chromeos-chrome-96.0.4664.208_rc-r1:  [ ok ]
chromeos-chrome-96.0.4664.208_rc-r1:  * Using python2.7 to build
chromeos-chrome-96.0.4664.208_rc-r1:  * Running stacked hooks for post_pkg_setup
chromeos-chrome-96.0.4664.208_rc-r1:  *    python_eclass_hack ...
chromeos-chrome-96.0.4664.208_rc-r1:  [ ok ]
chromeos-chrome-96.0.4664.208_rc-r1: 14:16:09 >>> Unpacking source...
chromeos-chrome-96.0.4664.208_rc-r1: 
chromeos-chrome-96.0.4664.208_rc-r1:  * If you want to develop or hack on the browser itself, you should follow the
chromeos-chrome-96.0.4664.208_rc-r1:  * simple chrome workflow instead of using emerge:
chromeos-chrome-96.0.4664.208_rc-r1:  * https://chromium.googlesource.com/chromiumos/docs/+/master/simple_chrome_workflow.md
chromeos-chrome-96.0.4664.208_rc-r1: 
chromeos-chrome-96.0.4664.208_rc-r1:  * Setting up CrOS credentials
chromeos-chrome-96.0.4664.208_rc-r1: '/home/ubuntu/.ssh' -> '/build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/homedir/.ssh'
chromeos-chrome-96.0.4664.208_rc-r1: '/home/ubuntu/.ssh/config' -> '/build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/homedir/.ssh/config'
chromeos-chrome-96.0.4664.208_rc-r1: '/home/ubuntu/.ssh/known_hosts' -> '/build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/homedir/.ssh/known_hosts'
chromeos-chrome-96.0.4664.208_rc-r1: '/home/ubuntu/.ssh/config.orig' -> '/build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/homedir/.ssh/config.orig'
chromeos-chrome-96.0.4664.208_rc-r1:  * CHROME_ORIGIN VALUE is LOCAL_SOURCE
chromeos-chrome-96.0.4664.208_rc-r1: Building Chromium with additional media codecs and containers.
chromeos-chrome-96.0.4664.208_rc-r1: 14:16:09 >>> Source unpacked in /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/work
chromeos-chrome-96.0.4664.208_rc-r1:  * Running stacked hooks for post_src_unpack
chromeos-chrome-96.0.4664.208_rc-r1:  *    asan_init ...
chromeos-chrome-96.0.4664.208_rc-r1:  [ ok ]
chromeos-chrome-96.0.4664.208_rc-r1:  * Running stacked hooks for pre_src_prepare
chromeos-chrome-96.0.4664.208_rc-r1:  *    build_toolchain_catch ...
chromeos-chrome-96.0.4664.208_rc-r1:  [ ok ]
chromeos-chrome-96.0.4664.208_rc-r1:  *    enable_cxx_exceptions ...
chromeos-chrome-96.0.4664.208_rc-r1:  [ ok ]
chromeos-chrome-96.0.4664.208_rc-r1: 14:16:09 >>> Preparing source in /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/work/chromeos-chrome-96.0.4664.208_rc ...
chromeos-chrome-96.0.4664.208_rc-r1:  * /home/ubuntu/chrome_root should be set here properly
chromeos-chrome-96.0.4664.208_rc-r1: rm: cannot remove 'c': Read-only file system
chromeos-chrome-96.0.4664.208_rc-r1:  * ERROR: chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1::cros-pi failed (prepare phase):
chromeos-chrome-96.0.4664.208_rc-r1:  *   Could not remove symlink
chromeos-chrome-96.0.4664.208_rc-r1:  * 
chromeos-chrome-96.0.4664.208_rc-r1:  * Call stack:
chromeos-chrome-96.0.4664.208_rc-r1:  *               ebuild.sh, line  125:  Called src_prepare
chromeos-chrome-96.0.4664.208_rc-r1:  *             environment, line 5327:  Called die
chromeos-chrome-96.0.4664.208_rc-r1:  * The specific snippet of code:
chromeos-chrome-96.0.4664.208_rc-r1:  *           rm -rf "${BUILD_OUT_SYM}" || die "Could not remove symlink";
chromeos-chrome-96.0.4664.208_rc-r1:  * 
chromeos-chrome-96.0.4664.208_rc-r1:  * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-96.0.4664.208_rc-r1:20230331-061607.log
chromeos-chrome-96.0.4664.208_rc-r1:  * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/temp/build.log
chromeos-chrome-96.0.4664.208_rc-r1:  * CWD: /home/ubuntu/chrome_root/src
chromeos-chrome-96.0.4664.208_rc-r1:  * S:   /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/work/chromeos-chrome-96.0.4664.208_rc
>>> 14:16:12 Installing (1 of 14) chromeos-base/chromeos-ssh-testkeys-0.0.1-r2::chromiumos to /build/rpi4/
>>> 14:16:16 Completed (1 of 14) chromeos-base/chromeos-ssh-testkeys-0.0.1-r2::chromiumos to /build/rpi4/
>>> 14:16:16 Jobs: 1 of 14 complete, 1 failed                Load avg: 1.81, 2.72, 3.16

 * Messages for package chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1 merged to /build/rpi4/:
[... cut ...]
 *             environment, line 5327:  Called die
 * The specific snippet of code:
 *           rm -rf "${BUILD_OUT_SYM}" || die "Could not remove symlink";
 * 
 * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-96.0.4664.208_rc-r1:20230331-061607.log
 * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/temp/build.log
 * CWD: /home/ubuntu/chrome_root/src
 * S:   /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/work/chromeos-chrome-96.0.4664.208_rc

nirocfz, 2023-03-31:
It can be seen that within the chroot environment, the current user is unable to modify any content within the chromium source code directory /home/ubuntu/chrome_root/src, resulting in a compilation failure. Please check the ownership and permissions of the source code directory.

wwd-615, 2023-04-01:
thank you very much

wwd-615, 2023-04-02:
I have encountered a new problem, can you help me to have a look?
chromeos-chrome-96.0.4664.208_rc-r1: ninja: build stopped: subcommand failed.
chromeos-chrome-96.0.4664.208_rc-r1:  * ERROR: chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1::cros-pi failed (compile phase):
chromeos-chrome-96.0.4664.208_rc-r1:  *   (no error message)
chromeos-chrome-96.0.4664.208_rc-r1:  * 
chromeos-chrome-96.0.4664.208_rc-r1:  * Call stack:
chromeos-chrome-96.0.4664.208_rc-r1:  *               ebuild.sh, line  125:  Called src_compile
chromeos-chrome-96.0.4664.208_rc-r1:  *             environment, line 5326:  Called chrome_make 'chrome' 'capture_unittests' 'dawn_end2end_tests' 'dawn_unittests' 'gl_tests' 'jpeg_decode_accelerator_unittest' 'ozone_gl_unittests' 'sandbox_linux_unittests' 'wayland_client_perftests' 'ppapi/examples/video_decode' 'bitmaptools' 'clear_system_cache' 'minidump_stackwalk' 'chromedriver' 'nacl_helper_bootstrap' 'nacl_helper'
chromeos-chrome-96.0.4664.208_rc-r1:  *             environment, line 1815:  Called die
chromeos-chrome-96.0.4664.208_rc-r1:  * The specific snippet of code:
chromeos-chrome-96.0.4664.208_rc-r1:  *       [[ "${ret}" -eq 0 ]] || die;
chromeos-chrome-96.0.4664.208_rc-r1:  * 
chromeos-chrome-96.0.4664.208_rc-r1:  * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-96.0.4664.208_rc-r1:20230402-144551.log
chromeos-chrome-96.0.4664.208_rc-r1:  * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/temp/build.log
chromeos-chrome-96.0.4664.208_rc-r1:  * CWD: /home/wwd/chrome_root/src
chromeos-chrome-96.0.4664.208_rc-r1:  * S:   /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/work/chromeos-chrome-96.0.4664.208_rc

nirocfz, 2023-04-03:
The common cause of error is forgetting to execute `gclient sync` in the Chromium source code root directory. Please try executing `gclient sync` and `emerge` it again.

wwd-615, 2023-04-03:
Ok, thank you very much. I'll try it

wwd-615, 2023-04-03:
I executed the gclient sync command, but still got the following error：
chromeos-chrome-96.0.4664.208_rc-r1: ninja: build stopped: subcommand failed.
chromeos-chrome-96.0.4664.208_rc-r1:  * ERROR: chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1::cros-pi failed (compile phase):
chromeos-chrome-96.0.4664.208_rc-r1:  *   (no error message)
chromeos-chrome-96.0.4664.208_rc-r1:  * 
chromeos-chrome-96.0.4664.208_rc-r1:  * Call stack:
chromeos-chrome-96.0.4664.208_rc-r1:  *               ebuild.sh, line  125:  Called src_compile
chromeos-chrome-96.0.4664.208_rc-r1:  *             environment, line 5326:  Called chrome_make 'chrome' 'capture_unittests' 'dawn_end2end_tests' 'dawn_unittests' 'gl_tests' 'jpeg_decode_accelerator_unittest' 'ozone_gl_unittests' 'sandbox_linux_unittests' 'wayland_client_perftests' 'ppapi/examples/video_decode' 'bitmaptools' 'clear_system_cache' 'minidump_stackwalk' 'chromedriver' 'nacl_helper_bootstrap' 'nacl_helper'
chromeos-chrome-96.0.4664.208_rc-r1:  *             environment, line 1815:  Called die
chromeos-chrome-96.0.4664.208_rc-r1:  * The specific snippet of code:
chromeos-chrome-96.0.4664.208_rc-r1:  *       [[ "${ret}" -eq 0 ]] || die;
chromeos-chrome-96.0.4664.208_rc-r1:  * 
chromeos-chrome-96.0.4664.208_rc-r1:  * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-96.0.4664.208_rc-r1:20230403-005955.log
chromeos-chrome-96.0.4664.208_rc-r1:  * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/temp/build.log
chromeos-chrome-96.0.4664.208_rc-r1:  * CWD: /home/wwd/chrome_root/src
chromeos-chrome-96.0.4664.208_rc-r1:  * S:   /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/work/chromeos-chrome-96.0.4664.208_rc
>>> 09:09:03 Jobs: 1 of 14 complete, 1 failed                Load avg: 14.9, 13.9, 7.1

 * Messages for package chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1 merged to /build/rpi4/:
 * Log file: /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-96.0.4664.208_rc-r1:20230403-005955.log

 * If you want to develop or hack on the browser itself, you should follow the
 * simple chrome workflow instead of using emerge:
 * https://chromium.googlesource.com/chromiumos/docs/+/master/simple_chrome_workflow.md
 * CHROME_ORIGIN VALUE is LOCAL_SOURCE
 * /home/wwd/chrome_root should be set here properly
 * ERROR: chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1::cros-pi failed (compile phase):
 *   (no error message)
 * 
 * Call stack:
 *               ebuild.sh, line  125:  Called src_compile
 *             environment, line 5326:  Called chrome_make 'chrome' 'capture_unittests' 'dawn_end2end_tests' 'dawn_unittests' 'gl_tests' 'jpeg_decode_accelerator_unittest' 'ozone_gl_unittests' 'sandbox_linux_unittests' 'wayland_client_perftests' 'ppapi/examples/video_decode' 'bitmaptools' 'clear_system_cache' 'minidump_stackwalk' 'chromedriver' 'nacl_helper_bootstrap' 'nacl_helper'
 *             environment, line 1815:  Called die
 * The specific snippet of code:
 *       [[ "${ret}" -eq 0 ]] || die;
 * 
 * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-96.0.4664.208_rc-r1:20230403-005955.log
 * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/temp/build.log
 * CWD: /home/wwd/chrome_root/src
 * S:   /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-96.0.4664.208_rc-r1/work/chromeos-chrome-96.0.4664.208_rc

wwd-615, 2023-04-04:
My build environment is ubuntu18 and the python version is Python 3.8. 0. The python version in cros_sdk is python3.6.12，Is the above error related to the python version?

nirocfz, 2023-04-04:
I have not encountered such a problem before, and I don't think it's related to the python version. Please go back to the previous steps to ensure that each step is correct, maybe you should try to start over again.

wwd-615, 2023-04-04:
I have a question, can you help me to answer it？
In the branch r96-dev, "Git clone https://github.com/fydeos/chromium_os-raspberry_pi.git." whether you need this command to "git clone - b r96-dev https://github.com/fydeos/chromium_os-raspberry_pi.git "?

fydeos-alpha, 2023-04-04:
> I have a question, can you help me to answer it？ In the branch r96-dev, "Git clone https://github.com/fydeos/chromium_os-raspberry_pi.git." whether you need this command to "git clone - b r96-dev https://github.com/fydeos/chromium_os-raspberry_pi.git "?

If you want to specify a branch in the git clone command, yes, you do need `-b` flag to tell git, otherwise it will default to the `main` branch. This is a basic git usage question, and it is beyond this project's scope. You need to familiarise yourself with the basics before attempting to do advanced stuff, like building Chromium OS.

As @nirocfz mentioned above, you have been using an old branch of this project, which was no longer in maintenance for quite some time. I also suggest you start over with r102-dev or r108-dev, which we released not long ago. Also, your build environment ubuntu 1804 is also quite dated, I would upgrade that as well.

wwd-615, 2023-04-04:
> If you want to specify a branch in the git clone command, yes, you do need `-b` flag to tell git, otherwise it will default to the `main` branch. This is a basic git usage question, and it is beyond [...]

Thank you very much for your advice
Maybe I have a problem expressing myself，What I mean is that whichever branch corresponds to the README document uses the command "git clone https://github.com/fydeos/chromium_os-raspberry_pi.git .". Does the overlays of the main branch work with all versions of chromium os?

fydeos-alpha, 2023-04-04:
We don't know if the main branch of this project would work on all versions of the Chromium OS platform. At this point in time, the main branch is the bleeding edge branch we are using to submit new code for new features, and we are testing it with Chromium OS r108. If you want to use the main branch, then it's recommended to use Chromium OS [repo](https://chromium.googlesource.com/chromiumos/manifest.git/+/refs/heads/release-R108-15183.B) to match it.

wwd-615, 2023-04-06:
When I switch to the r102-dev branch, the following error still occurs, and I'm sure I executed the gclinet sync command

>>> 13:35:50 Failed to emerge chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1 for /build/rpi4/, Log file:
>>> 13:35:50   /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-102.0.5005.179_rc-r1:20230406-052617.log
>>> 13:35:50 === Start output for job chromeos-chrome-102.0.5005.179_rc-r1 ===
chromeos-chrome-102.0.5005.179_rc-r1:  * Package:    chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1
chromeos-chrome-102.0.5005.179_rc-r1:  * Repository: cros-pi
chromeos-chrome-102.0.5005.179_rc-r1:  * USE:        accessibility afdo_use arm autotest build_tests buildcheck cfi chrome_debug chrome_media chrome_remoting cros-debug debug_fission dwarf5 elibc_glibc fonts highdpi kernel_linux libcxx nacl neon oobe_config opengles orderfile_use ozone_platform_default_gbm ozone_platform_gbm ozone_platform_headless runhooks thinlto userland_GNU xkbcommon
chromeos-chrome-102.0.5005.179_rc-r1:  * FEATURES:   network-sandbox splitdebug userpriv
chromeos-chrome-102.0.5005.179_rc-r1:  * Running stacked hooks for pre_pkg_setup
chromeos-chrome-102.0.5005.179_rc-r1:  *    sysroot_build_bin_dir ...
chromeos-chrome-102.0.5005.179_rc-r1:  [ ok ]
chromeos-chrome-102.0.5005.179_rc-r1:  * Using python3.6 to build
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:18 >>> Unpacking source...
chromeos-chrome-102.0.5005.179_rc-r1: 
chromeos-chrome-102.0.5005.179_rc-r1:  * If you want to develop or hack on the browser itself, you should follow the
chromeos-chrome-102.0.5005.179_rc-r1:  * simple chrome workflow instead of using emerge:
chromeos-chrome-102.0.5005.179_rc-r1:  * https://chromium.googlesource.com/chromiumos/docs/+/HEAD/simple_chrome_workflow.md
chromeos-chrome-102.0.5005.179_rc-r1:  * 
chromeos-chrome-102.0.5005.179_rc-r1:  * Otherwise, if you have a Chrome checkout already present on your machine, you can
chromeos-chrome-102.0.5005.179_rc-r1:  * re-enter the cros_sdk with the --chrome-root arg and set CHROME_ORIGIN to
chromeos-chrome-102.0.5005.179_rc-r1:  * LOCAL_SOURCE. This will bypass the lengthy sync_chrome phase here.
chromeos-chrome-102.0.5005.179_rc-r1:  * Note: when using --chrome-root, the ebuild won't update or modify the Chrome checkout.
chromeos-chrome-102.0.5005.179_rc-r1:  * So make sure it's at a compatible Chrome version before proceeding.
chromeos-chrome-102.0.5005.179_rc-r1: 
chromeos-chrome-102.0.5005.179_rc-r1:  * Setting up CrOS credentials
chromeos-chrome-102.0.5005.179_rc-r1: '/home/wwd/.ssh' -> '/build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/homedir/.ssh'
chromeos-chrome-102.0.5005.179_rc-r1: '/home/wwd/.ssh/config' -> '/build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/homedir/.ssh/config'
chromeos-chrome-102.0.5005.179_rc-r1: '/home/wwd/.ssh/known_hosts' -> '/build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/homedir/.ssh/known_hosts'
chromeos-chrome-102.0.5005.179_rc-r1: '/home/wwd/.ssh/config.orig' -> '/build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/homedir/.ssh/config.orig'
chromeos-chrome-102.0.5005.179_rc-r1:  * CHROME_ORIGIN VALUE is LOCAL_SOURCE
chromeos-chrome-102.0.5005.179_rc-r1: Building Chromium with additional media codecs and containers.
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:18 >>> Source unpacked in /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/work
chromeos-chrome-102.0.5005.179_rc-r1:  * Running stacked hooks for post_src_unpack
chromeos-chrome-102.0.5005.179_rc-r1:  *    asan_init ...
chromeos-chrome-102.0.5005.179_rc-r1:  [ ok ]
chromeos-chrome-102.0.5005.179_rc-r1:  * Running stacked hooks for pre_src_prepare
chromeos-chrome-102.0.5005.179_rc-r1:  *    build_toolchain_catch ...
chromeos-chrome-102.0.5005.179_rc-r1:  [ ok ]
chromeos-chrome-102.0.5005.179_rc-r1:  *    enable_cxx_exceptions ...
chromeos-chrome-102.0.5005.179_rc-r1:  [ ok ]
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:18 >>> Preparing source in /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/work/chromeos-chrome-102.0.5005.179_rc ...
chromeos-chrome-102.0.5005.179_rc-r1:  * /home/wwd/chrome_root should be set here properly
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:18 >>> Source prepared.
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:18 >>> Configuring source in /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/work/chromeos-chrome-102.0.5005.179_rc ...
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:19  * Appending a library link instruction (-lmmal); libraries to link to should not be passed through LDFLAGS
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:19  * Appending a library link instruction (-lmmal_core); libraries to link to should not be passed through LDFLAGS
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:19  * Appending a library link instruction (-lmmal_util); libraries to link to should not be passed through LDFLAGS
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:19  * Appending a library link instruction (-lmmal_vc_client); libraries to link to should not be passed through LDFLAGS
chromeos-chrome-102.0.5005.179_rc-r1: 13:26:19  * Appending a library link instruction (-lbcm_host); libraries to link to should not be passed through LDFLAGS
chromeos-chrome-102.0.5005.179_rc-r1:  * Building with the compiler settings:
[... cut ...]
chromeos-chrome-102.0.5005.179_rc-r1:  * Build log: /build/rpi4/tmp/portage/logs/chromeos-base:chromeos-chrome-102.0.5005.179_rc-r1:20230406-052617.log
chromeos-chrome-102.0.5005.179_rc-r1:  * Stable log symlink: /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/temp/build.log
chromeos-chrome-102.0.5005.179_rc-r1:  * CWD: /home/wwd/chrome_root/src
chromeos-chrome-102.0.5005.179_rc-r1:  * S:   /build/rpi4/tmp/portage/chromeos-base/chromeos-chrome-102.0.5005.179_rc-r1/work/chromeos-chrome-102.0.5005.179_rc
>>> 13:35:50 Jobs: 4 of 12 complete, 1 failed                Load avg: 2.24, 1.66, 1.40

nirocfz, 2023-04-06:
It is possible that the compilation process was terminated due to insufficient system resources. You can try to find out if there is any relevant information when the error occurs by looking in `/var/log/messages` (or perhaps `/var/log/syslog`).

wwd-615, 2023-04-06:
Thank you very much, I tried to give the virtual machine more memory. Another question I would like to ask you is whether you can directly use chromiumos system without chromium compilation

wwd-615, 2023-04-06:
Due to memory problems on my computer, I want not building  copy of Chromium, How to operate, can help me?

wwd-615, 2023-04-07:
thank you very much.

closed 2023-04-07 by wwd-615