FydeOS/chromium_os-raspberry_pi #245: System file size allocation too large. Unable to load linux environment or download apps

Namrepus221, 2023-04-17:
I'm trying to run ChromiumOS with a Raspberry Pi and have a 256gb SD Card. When I attempt to load the Linux environment it immediately fails because I have a System file of 253gb and only 3 GB of "free" space available.

How do I shrink the size of the system file when it's written so I don't use 99% of it on useless stuff?

Namrepus221, 2023-04-17:
[image]

Damenly, 2023-04-17:
It's a known bug which will be fixed in next release. 
For current version, you can slove it in:
1. back up your data first 
2. follow https://faq.fydeos.com/en/getting-started/shell-access/
3. call `mount -oremount,rw /`
4. call `echo "echo yes" > /usr/sbin/is_running_from_installer`
5. call `chmod +x /usr/sbin/is_running_from_installer`
6. call `/usr/sbin/expand-partition.sh -d $(findmnt -n -o source /mnt/stateful_partition)`
7. reboot

closed 2023-05-06 by fydeos-alpha