FydeOS/chromium_os-raspberry_pi #246: Can't Enable Developer Mode

CommanderGL, 2023-04-22:
**Describe the bug or issue**
Can't Enable Developer Mode

**Describe your setup**
 - Raspberry Pi 4 Model B 4GB
 - Version 92.0.4515.183 (Official Build)
 - Booting off USB. Mouse plugged into keyboard. Keyboard plugged into pi.

**Expected behaviour**
Special shortcut to enable without power button.

.

**Additional context**
How do you do it without power button.

fydeos-alpha, 2023-04-23:
what "developer mode" did you mean, could you elaborate what would you like to achieve under this developer mode?

CommanderGL, 2023-04-23:
I'm trying to run APKs but the only way to enable developer mode is to hold esc, refresh (f3), and power (Which my keyboard doesn't have.)

fydeos-alpha, 2023-04-23:
Chromium OS for Raspberry Pi is a Chromium OS build, which does not support the Android subsystem. Read the TL;DR in this project for details.

If you want to use the Android subsystem, you need to use FydeOS for You - Raspberry Pi, available to download at https://fydeos.io/download/device/rpi4-fydeos

With FydeOS, you don't need to enable "developer mode" to sideload apk files.

Support for FydeOS is not in the scope of this project; try the [Discord server](https://discord.gg/Qv3kxGeMJQ) or [Telegram Group](https://t.me/hi_fydeos).

closed 2023-04-23 by fydeos-alpha