FydeOS/chromium_os-raspberry_pi #249: [Issue report] Removing the OS from my sdcard

SSP6904, 2023-06-26:
**Describe the bug or issue**
I was going to remove the OS from the SD card, but attempts to do so causes the partition table to revert the changes unexpectedly, including the use of erasing and/or formatting some partitions. I also noticed that when using the OS and doing changes on their makes it revert and clear my progress, even using powerwash to format it as well..

**Describe your setup**
 - Raspberry Pi 4B 8GB
 - chromiumos image r89r1
 - Keyboard and mouse

**To Reproduce**
Steps to reproduce the behavior:
Not able to do

**Expected behaviour**
I expected the drive to erase it.

**Screenshots and/or hardware setup photos**
If applicable, add screenshots to help explain your problem. If your problem is hardware related, it's much easier to include a photo of how your setup looks like.

Damenly, 2023-06-27:
Sorry I am not a fan of GUI gparted. You can try commmd line:

1. `sudo fdisk /dev/sda`
2. Press key `g`  and confirm 
3. Press key w 
4. `lsblk`

SSP6904, 2023-06-27:
Didn't work, same thing

Damenly, 2023-06-27:
Please paste all outout or screen record.

SSP6904, 2023-06-27:
[image]

I did what you told me to type, but the partitions came back, also on every reboot.

Damenly, 2023-06-27:
The fdisk reported that your SD card was mounted.

0. lsblk
1. sudo umount /dev/sda*
2.sudo fdisk /dev/sda
3. Press key `g` and confirm using key `enter`
4. Press key w
5. lsblk

closed 2023-06-28 by Damenly

SSP6904, 2023-09-18:
That also did not work. I haven't responded in a while so I can't fully remember part of this issue.