FydeOS/chromium_os-raspberry_pi #250: [Issue report]

maroonmoon13, 2023-07-18:
**Describe the bug or issue**
Display corruption on rpi4 with the official raspberry pi touchscreen.
**Describe your setup**
 - Rpi 4B with 2GB RAM
 - [chromiumos_image_r108-rpi4b.img.xz](https://github.com/FydeOS/chromium_os-raspberry_pi/releases/download/r108/chromiumos_image_r108-rpi4b.img.xz)
 - Rpi4B connected to the Raspberry pi touchscreen, as seen here: https://thepihut.com/products/official-raspberry-pi-7-touchscreen-display

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the image on the SD card with either Balenaetcher or rpi imager
2. With the LCD touchscreen connected, boot pi
3. The LCD is all white and black strips, Cannot say if the device did boot or hung as there is no information on the screen.

**Expected behaviour**
Proper boot

**Additional context**
I verified that the LCD works fine with the official Raspberry Pi Linux distribution. ALso, I was able to boot of an old version of chromium OS. Unfortunately, I did not note the version before I flashed the new chromium version on the SD card.

fydeos-alpha, 2023-07-18:
Try this: https://fydeos.io/docs/knowledge-base/recipes/enable-fake-kms-for-raspberry-Pi

maroonmoon13, 2023-07-18:
Thank you!
 
I tried the second method of modifying the image on Ubuntu 22.04. I had trouble with the command:
`sudo mount -t msdos /dev/mapper/loop0p12 -o rw /mnt/crosp12
`
the command which worked was 
`sudo mount -t vfat /dev/mapper/loop0p12 -o rw /mnt/crosp12
`

Also, I had to install the package kpartx, of course :)

I will test and update if the issue still persists.

closed 2023-11-20 by maroonmoon13