FydeOS/chromium_os-raspberry_pi #251: Doc doesn't work - https://github.com/FydeOS/chromium_os-raspberry_pi#system-requirement

zieglerjm, 2023-08-17:
In the "Fetch Chromium OS source code" section it has:

`repo init -u https://chromium.googlesource.com/chromiumos/manifest.git --repo-url https://chromium.googlesource.com/external/repo.git -b release-R102-14695.B`

Results in the following on Ubuntu 18.04.6:

`  File "/home/{username}/cros-pi/.repo/repo/main.py", line 81
    file=sys.stderr,
        ^
SyntaxError: invalid syntax`

Is there a know working "-b" options that the page can be upgraded to?

fydeos-alpha, 2023-08-18:
Here is the man page: https://source.android.com/docs/setup/create/repo

Where did you get your `repo` from? In the context of Chromium OS development, it's recommended to follow https://commondatastorage.googleapis.com/chrome-infra-docs/flat/depot_tools/docs/html/depot_tools_tutorial.html#_setting_up

zieglerjm, 2023-08-19:
On Ubuntu 20.04.06 I got a lot farther.  Running "./build_packages --board=rpi4 --nowithautotest" got the below error - 

(dependency required by "virtual/target-os" [argument])
20:24:23 ERROR   : Fri Aug 18 08:24:23 PM EDT 2023
20:24:23 ERROR   :    PGID    PPID     PID     ELAPSED     TIME %CPU COMMAND
20:24:23 ERROR   :  385153       4  385153       13:13 00:00:00  0.0 -bash
20:24:23 ERROR   :  443309  385153  443309       02:13 00:00:00  0.4  \_ python3 /mnt/host/source/chromite/bin/build_packages --board=rpi4 --nowithautotest
20:24:23 ERROR   :  443309  443309  454021       00:42 00:00:00  0.0      \_ bash /mnt/host/source/src/scripts/build_packages.sh --script-is-run-only-by-chromite-and-not-users --board rpi4 --board_root /build/rpi4 --jobs 8 --nowithautotest
20:24:23 ERROR   :  443309  454021  454292       00:00 00:00:00  0.0          \_ bash /mnt/host/source/src/scripts/build_packages.sh --script-is-run-only-by-chromite-and-not-users --board rpi4 --board_root /build/rpi4 --jobs 8 --nowithautotest
20:24:23 ERROR   :  443309  454292  454293       00:00 00:00:00  0.0              \_ ps f -o pgid,ppid,pid,etime,cputime,%cpu,command
20:24:23 ERROR   : Arguments of 454021:  '--script-is-run-only-by-chromite-and-not-users' '--board' 'rpi4' '--board_root' '/build/rpi4' '--jobs' '8' '--nowithautotest'
20:24:23 ERROR   : Backtrace:  (most recent call is last)
20:24:23 ERROR   :  build_packages.sh:395:main(), called: die_err_trap
20:24:23 ERROR   :
20:24:23 ERROR   : Command failed:
20:24:23 ERROR   :   Command '( if [[ "${FLAGS_run_remoteexec}" -eq "${FLAGS_TRUE}" ]]; then
20:24:23 ERROR   :     info "Starting RBE reproxy."; bootstrap="${RECLIENT_DIR}/bootstrap --cfg=${REPROXY_CFG}       --re_proxy=${RECLIENT_DIR}/reproxy"; ${bootstrap}; trap "${bootstrap} --shutdown" EXIT;
20:24:23 ERROR   : else
20:24:23 ERROR   :     if [[ "${FLAGS_run_goma}" -eq "${FLAGS_TRUE}" ]]; then
20:24:23 ERROR   :         info "Starting goma compiler_proxy."; goma_ctl="${GOMA_DIR:-${HOME}/goma}/goma_ctl.py"; "${goma_ctl}" restart; trap "'${goma_ctl}' stop" EXIT;
20:24:23 ERROR   :     fi;
20:24:23 ERROR   : fi; info_run sudo -E "${EMERGE_CMD[@]}" "${EMERGE_FLAGS[@]}" "${PACKAGES[@]}" --useoldpkg-atoms="${CRITICAL_SDK_PACKAGES[*]}" --rebuild-exclude="${CRITICAL_SDK_PACKAGES[*]}" )' exited with nonzero code: 1
20:24:23.212: NOTICE: Tree Status: 🌲 is open: b/296510507 should be fixed
20:24:23.212: ERROR: return code: 1; command: bash PosixPath('/mnt/host/source/src/scripts/build_packages.sh') --script-is-run-only-by-chromite-and-not-users --board rpi4 --board_root /build/rpi4 --jobs 8 --nowithautotest
return code: 1; command: bash PosixPath('/mnt/host/source/src/scripts/build_packages.sh') --script-is-run-only-by-chromite-and-not-users --board rpi4 --board_root /build/rpi4 --jobs 8 --nowithautotest
cmd=['bash', PosixPath('/mnt/host/source/src/scripts/build_packages.sh'), '--script-is-run-only-by-chromite-and-not-users', '--board', 'rpi4', '--board_root', '/build/rpi4', '--jobs', '8', '--nowithautotest'], extra env={'CROS_METRICS_DIR': '/tmp/tmpj3erhet6'}

[image]

fydeos-alpha, 2023-12-21:
We have updated the overlay file structure and tested the build sequence should now work.

closed 2023-12-21 by fydeos-alpha