FydeOS/chromium_os-raspberry_pi #253: [Issue report] Failed to determine device ID (Enterprise Enrollment)

KyGuy2002, 2024-02-03:
**Describe the bug or issue**
When trying to enroll this device into an enterprise, it gives the error "Oops! The system failed to determine device identifiers for this device."

**Describe your setup**
 - Raspberry Pi Model 4b
 - r114-r1-rpi4b
 - Logitech wireless (usb dongle) keyboard, Logitech wired mouse, and Acer touchscreen monitor.

**To Reproduce**
Steps to reproduce the behavior:
1. Click "Enterprise Enrollment" at the bottom right.
2. Enter email and password.
3. Observe error.

**Expected behaviour**
The enterprise enrollment should succeed.

**Additional context**
I saw the other issues, but they all said this was fixed when VPD was added in r92-r2.  I also tried rebooting, logging in as a guest and viewing the SN in chrome://system, and using the r92-r2 release.  Is there something else I need to try?  Some tutorials for chromebooks said to go into dev mode, but there is no dev mode on this so I'm not sure what to do.  Thanks!

nirocfz, 2024-02-05:
Now, this project retains the minimal changes to the original ChromiumOS code. You can try using the [openFyde version](https://github.com/openFyde/overlay-rpi4-openfyde/releases/tag/r114-r1), which includes our modifications for VPD.

KyGuy2002, 2024-02-06:
I just tried using openFyde and im experiencing the same issue.

Thank you.

nirocfz, 2024-02-06:
> I just tried using openFyde and im experiencing the same issue.
> 
> Thank you.

Then this issue is likely unrelated to VPD. Please try adding `--allow-init-device-policy-without-state-keys` to `/etc/chrome_dev.conf`, then restart the device and try again.

KyGuy2002, 2024-02-06:
How would I add this?  Do I need to login in guest mode and put this somewhere?

fydeos-alpha, 2024-02-06:
You will need to gain shell access (by pressing ctrl + alt + F2 this should get you into tty), login with `chronos`, you can change to root privilege by `sudo -i`. Also do a `mount -o remount rw /` to unlock the root file system write permission. Then you may edit the /etc/chrome_dev.conf with any editor available.

Basic command line knowledge and skills are assumed to perform the above.

KyGuy2002, 2024-02-06:
Thanks, this worked!

closed 2024-02-06 by KyGuy2002