FydeOS/chromium_os-raspberry_pi #254: [Issue report] Flashing/Flickering and Glitchy Cursor

KyGuy2002, 2024-02-06:
**Describe the bug or issue**
All popup dialogs constantly flash and flicker, kiosk mode browser never even renders, just flashes white and black then back to "Preparing app profile...", and text everywhere (even in the CTRL+ALT+F2 console) flashes and freezes, requiring cursor movements to update.

**Describe your setup**
 - Raspberry Pi Model 4b
 - r114-r1-rpi4b
 - Logitech wireless (usb dongle) keyboard, Logitech wired mouse, and Acer touchscreen monitor.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the os in any way past the login screen

**Expected behaviour**
It should not flicker.

**Additional context**
This issue #69 describes it pretty perfectly.  I also tried [this](https://github.com/FydeOS/chromium_os-raspberry_pi/issues/53#issuecomment-784899372) but didn't fix anything.

Thanks!

fydeos-alpha, 2024-02-06:
What about this: https://fydeos.io/docs/knowledge-base/recipes/enable-fake-kms-for-raspberry-Pi

Does this help?

KyGuy2002, 2024-02-06:
Unfortunately that didn't make a difference.

fydeos-alpha, 2024-02-06:
We don't seem to be able to reproduce this issue.

Evidence:
[pi4.webm](https://github.com/FydeOS/chromium_os-raspberry_pi/assets/37786989/6070dc84-af8c-4a1b-89b2-0f5831e44008)

The setup:
[image]

- Pi 4B, 8GB RAM
- USB HDD via USB
- Mouse / Keyboard via USB
- Micro HDMI to 2560x1440 monitor, 59.95Hz
- also works at 4k 30Hz

fydeos-alpha, 2024-02-06:
( it's known that the graphics have some glitches in guest mode. But you did mention that you "signed in", so I'd believe it's something different. Guest mode issues will be fixed in the next release

KyGuy2002, 2024-02-07:
I just tested, and I am not having these issues when signed into my google account.  I AM experiencing them in guest mode like you said, and also in enterprise enrolled kiosk mode.  As stated the kiosk app wont even ever render, just flickers between gray and white screens every few seconds.

I realize now that I have been using openFyde (because the enterprise enrollment only works there #253), so sorry for posting in the wrong place.  Should I create a new issue on that repo?

fydeos-alpha, 2024-02-07:
Yep this proves that the kiosk mode and guest mode share the same (userless) display mechanism so the issue was inherited. We know what went wrong and will fix this in the coming release. It's already in our internal bug tracker so you don't need to create a new one, stay tuned for the next release please.

KyGuy2002, 2024-02-07:
Thats what I thought.  I will wait for the next release of openFyde, do you have an estimate of when this will be available? (few days, few weeks, etc?)  Thanks again!

fydeos-alpha, 2024-02-07:
It'd be in weeks, we hope. We'll do our best. Leave this open and I will close it once the issue is resolved and you are happy

KyGuy2002, 2024-02-07:
Sounds good.  Thank you for being so responsive and helpful!